ros2/ros2cli #6: Can't tell daemon which RMW implementation to use

dhood, 2017-06-16:
I am debugging something and want to switch rmw implementation for the new CLI tools.

I noticed `RMW_IMPLEMENTATION=rmw_connext_cpp ros2 node list` was not outputting the connext license. I figured it was because of https://github.com/ros2/rmw_implementation/issues/23 and that it was using fastrtps underneath.

However, I _have_ built with connext support, and even the following doesn't output the connext license:
```
RMW_IMPLEMENTATION=rmw_connext_cpp RCL_ASSERT_RMW_ID_MATCHES=rmw_connext_cpp ros2 node list
```

@wjwwood told me this is because the daemon is started once and then not communicated to using DDS, so the assertion is never checked anywhere.

Ideally I would expect `RMW_IMPLEMENTATION=rmw_connext_cpp ros2 node list` to use connext. But at the very least I would expect it to not ignore the `RCL_ASSERT_RMW_ID_MATCHES=rmw_connext_cpp` environment variable.

I understand why this happens, it's just misleading as a user expecting to have control over those things like for other participants.

dirk-thomas, 2017-06-16:
Since you assigned the ticket to yourself how do you plan to address the issue?

dhood, 2017-06-16:
it was just by habit. I unassigned but it doesn't look like it updated.

dirk-thomas, 2017-06-19:
I hope #11 addresses this problem sufficiently.

closed 2017-06-21 by dirk-thomas