ros2/ros2cli #8: TypeError if there are no topics being published

codebot, 2017-06-17:
I love this program :grinning: !! it's awesome!  But maybe I'm doing something wrong, but if nobody is publishing anything and I type `ros2 topic list`, I get this:

```
mquigley@feather2:~/ros2$ ros2 topic list 
Traceback (most recent call last):
  File "/home/mquigley/ros2/install/bin/ros2", line 9, in <module>
    load_entry_point('ros2cli', 'console_scripts', 'ros2')()
  File "/home/mquigley/ros2/build/ros2cli/ros2cli/cli.py", line 64, in main
    rc = extension.main(parser=parser, args=args)
  File "/home/mquigley/ros2/build/ros2topic/ros2topic/command/topic.py", line 38, in main
    return extension.main(args=args)
  File "/home/mquigley/ros2/build/ros2topic/ros2topic/verb/list.py", line 41, in main
    no_demangle=args.no_demangle)
  File "/home/mquigley/ros2/build/ros2topic/ros2topic/api/__init__.py", line 20, in get_topic_names_and_types
    topic_names_and_types = node.get_topic_names_and_types(no_demangle=no_demangle)
TypeError: __call__() got an unexpected keyword argument 'no_demangle'
```

codebot, 2017-06-17:
sorry my issue report wasn't very accurate. It looks like regardless of whether there are topics present or not, the initial invocation of `ros2 topic list` gives the correct answer, but a repeated invocation fails with the backtrace listed above. If you `pkill _ros2_daemon` and run `ros2 topic list` again, it gives the right answer, but subsequent invocations (before killing `_ros2_daemon`) will fail.

(That being said, it's very possible that I've got something messed up in my environment.)

wjwwood, 2017-06-17:
That's my bad. I noticed this problem, today but in the chaos I forgot to fix it. I don't know how to fix it properly, but I can remove the feature that's breaking it. I'll open a pr, asap.

wjwwood, 2017-06-17:
You can also kill the daemon and then permanently disable it with this:

```diff
diff --git a/ros2cli/ros2cli/node/strategy.py b/ros2cli/ros2cli/node/strategy.py
index e1b4ed1..1403cbe 100644
--- a/ros2cli/ros2cli/node/strategy.py
+++ b/ros2cli/ros2cli/node/strategy.py
@@ -26,7 +26,7 @@ class NodeStrategy(object):
         if is_daemon_running(args):
             self.node = DaemonNode(args)
         else:
-            spawn_daemon(args)
+            # spawn_daemon(args)
             self.node = DirectNode(args)

     def __enter__(self):
```

As a workaround until I get it built on my desktop.

codebot, 2017-06-17:
No rush. It's late enough 🌃 that all this talk of commenting lines that spawn demons has me scared now 😨. Best to work on this code in the bright light of day ☀️

wjwwood, 2017-06-17:
Some of my best work happens at night 🦇: https://github.com/ros2/ros2cli/pull/9

closed 2017-06-17 by wjwwood