ros2/ros2cli #16: Can't pass arguments through to nodes with `ros2 run`

dhood, 2017-06-26:
Running on OS X from binaries, but I've also seen it on linux from source.

```
$ ./lib/demo_nodes_cpp/talker -t /chatter_qos
Publishing: 'Hello World: 1'
Publishing: 'Hello World: 2'
^Csignal_handler(2)
$ ros2 run demo_nodes_cpp talker -t /chatter_qos
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: unrecognized arguments: -t /chatter_qos
```

Also happens with python nodes:
```
$ ros2 run demo_nodes_py listener_qos -n 10
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: unrecognized arguments: -n 10
```

dirk-thomas, 2017-06-26:
Have you tried it with the approach described in the help text (https://github.com/ros2/ros2cli/blob/5fe3be6a28313a740019f9f9260047fd9f6fa4cf/ros2run/ros2run/command/run.py#L39-L41)?

dhood, 2017-06-26:
Thanks, that fixes it. user error!

closed 2017-06-26 by dhood