ros2/ros2cli #17: pyyaml dependency in ros2 topic echo

dhood, 2017-06-26:
I didn't have `pyyaml` installed and got this error:

```
$ ros2 run demos_nodes_cpp listener
Failed to load entry point 'echo': No module named 'yaml'
```

I am adding `python3 -m pip install pyyaml` to the installation instructions, but I noticed it's not listed in the setup.py for `ros2topic` where I think it should be. Is that correct?

Kukanani, 2017-06-26:
There's also `apt install python3-yaml`, which appears to satisfy the requirement as well.

I'm experiencing similar behavior on Windows, and I think I may need to use the `pip` install.

mikaelarguedas, 2017-06-26:
> I'm experiencing similar behavior on Windows, and I think I may need to use the pip install

:+1: we should use the apt version on linux and the pip version on the other platforms

dirk-thomas, 2017-06-26:
Added to https://github.com/ros2/ros2cli/pull/15/commits/41d411e93ed09b6742e60e5420fbfd86d66d2bbb

dhood, 2017-06-26:
How come it doesn't also belong in the setup.py?

dirk-thomas, 2017-06-26:
The manifest contains the dependency information. We don't have to duplicate them into the setup.py file.

wjwwood, 2017-06-26:
> The manifest contains the dependency information. We don't have to duplicate them into the setup.py file.

There are things that are not provided by pip (like `rclpy`) but it doesn't hurt to put things into the `setup.py` which can be. But for packages which only have other python dependencies, it would be good to have a functional `setup.py`. So I don't agree with this as a general rule.

dhood, 2017-06-27:
Seems especially strange that the setup.py currently lists `ros2cli` but not `pyyaml`.

Updating the instructions to install pyyaml from pip on windows and osx and python3-yaml from apt for linux. https://github.com/ros2/ci/pull/89 does the same for CI jobs

PR #15 (2017-06-27): various fixes and improvements
Beside other changes fixes #17.
closed 2017-06-27 by dirk-thomas