ros2/ros2cli #18: [ros2 topic pub] Cannot send strings containing single quotes

mikaelarguedas, 2017-06-26:
Without quotes: OK
```bash
$ ros2 topic pub /chatter std_msgs/String '{"data":"Hello World"}'
publisher: beginning loop
publishing std_msgs.msg.String(data='Hello World'
```
With escaped double quotes: OK
```bash
$ ros2 topic pub /chatter std_msgs/String '{"data":"Hello\"World"}'
publisher: beginning loop
publishing std_msgs.msg.String(data='Hello"World')
```

With unescaped single quotes: Doesn't work
```bash
$ ros2 topic pub /chatter std_msgs/String '{"data":"Hello'World"}'
> 
```

With unescaped single quotes: Doesn't work
```bash
$ ros2 topic pub /chatter std_msgs/String '{"data":"Hello\'World"}'
> 
```

dirk-thomas, 2017-06-26:
The correct way to escape this on the shell is:

```
ros2 topic pub /chatter std_msgs/String '{"data":"Hello'"'"'World"}'
```

Please see https://stackoverflow.com/questions/1250079/how-to-escape-single-quotes-within-single-quoted-strings for more information.

mikaelarguedas, 2017-06-26:
Sounds good, user error then

closed 2017-06-26 by mikaelarguedas