ros2/ros2cli #22: Issues using ros2 topic pub with Point message

dhood, 2017-06-27:
I am having a hard time publishing `PoseWithCovarianceStamped` messages for the AMCL demo in ROS 2. I thought I was doing something wrong with JSON formatting, but it happens even with https://github.com/ros2/ros2cli/pull/19 that switches to YAML parsing.

I've narrowed it down to a simpler problem: I can't publish `Point` messages.
Publishing the same message with ROS 1's `rostopic pub` works fine, using the autocompleted template from ROS 1. But it errors in ROS 2:

```
ros2 topic pub initialpose geometry_msgs/Point "x: 0.0                
y: 0.0
z: 0.0" 
Failed to load entry point 'pub': EOL while scanning string literal (pub.py, line 72)
usage: ros2 topic [-h] [--include-hidden-topics]
                  Call `ros2 topic <command> -h` for more detailed usage. ...
ros2 topic: error: argument Call `ros2 topic <command> -h` for more detailed usage.: invalid choice: 'pub' (choose from 'echo', 'list')
```

It must need to be formatted differently to ROS 1, but I have run out of ideas!

<details>
  <summary>Message I actually want to publish</summary>

There's still probably user error on my part but if someone has suggestions I'd appreciate it.

```
ros2 topic pub initialpose geometry_msgs/PoseWithCovarianceStamped "header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: 'map'   
pose:
  pose:
    position: {x: 5.5, y: 4.5, z: 0.0}
    orientation: {x: 0.0, y: 0.0, z: 0.0, w: 1.0}
  covariance: [0.5, 0.5, 0.0, 0.0, 0.0, 0.0, 0.0, 0.5, 0.5, 0.0, 0.0, 0.0, 0.0, 0.0,    0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0,
    0.0, 0.0, 0.0, 0.0, 0.0, 0.5]"
Traceback (most recent call last):
  File "/home/dhood/ros2_ws/install_isolated/ros2cli/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.0.0', 'console_scripts', 'ros2')()
  File "/home/dhood/ros2_ws/install_isolated/ros2cli/lib/python3.5/site-packages/ros2cli/cli.py", line 64, in main
    rc = extension.main(parser=parser, args=args)
  File "/home/dhood/ros2_ws/install_isolated/ros2topic/lib/python3.5/site-packages/ros2topic/command/topic.py", line 38, in main
    return extension.main(args=args)
  File "/home/dhood/ros2_ws/install_isolated/ros2topic/lib/python3.5/site-packages/ros2topic/verb/pub.py", line 43, in main
    return main(args)
  File "/home/dhood/ros2_ws/install_isolated/ros2topic/lib/python3.5/site-packages/ros2topic/verb/pub.py", line 47, in main
    publisher(args.message_type, args.topic_name, args.values)
  File "/home/dhood/ros2_ws/install_isolated/ros2topic/lib/python3.5/site-packages/ros2topic/verb/pub.py", line 69, in publisher
    setattr(msg, field_name, field_type(field_value))
TypeError: __init__() takes 1 positional argument but 2 were given
```
</details>

dirk-thomas, 2017-06-27:
Please try: `ros2 topic pub initialpose geometry_msgs/Point "{x: 1, y: 2, z: 3}"`

dhood, 2017-06-27:
ok, Point is working, thanks. I wonder if the issue is just with nested messages. Do you know of a way to get Header to work?
Neither of these work for me:
`ros2 topic pub topic_name std_msgs/Header "{stamp: {sec: 0, nanosec: 0}, frame_id: 'map'}"`
```
ros2 topic pub topic_name std_msgs/Header "stamp:
  sec: 0
  nanosec: 0
frame_id: 'map'"
```

dhood, 2017-06-27:
Here's a better comparison with ROS 1 (since header has changed in ROS 2):

ROS 1:
```
rostopic pub asdf geometry_msgs/Wrench "{force: {x: 0.0, y: 0.0, z: 0.0}, torque: {x: 0.0, y: 0.0, z: 0.0}}"
publishing and latching message. Press ctrl-C to terminate
```

ROS  2:
```
ros2 topic pub asdf geometry_msgs/Wrench "{force: {x: 0.0, y: 0.0, z: 0.0}, torque: {x: 0.0, y: 0.0, z: 0.0}}"
Traceback (most recent call last):
  File "/home/dhood/ros2_ws/install_isolated/ros2cli/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.0.0', 'console_scripts', 'ros2')()
  File "/home/dhood/ros2_ws/install_isolated/ros2cli/lib/python3.5/site-packages/ros2cli/cli.py", line 64, in main
    rc = extension.main(parser=parser, args=args)
  File "/home/dhood/ros2_ws/install_isolated/ros2topic/lib/python3.5/site-packages/ros2topic/command/topic.py", line 38, in main
    return extension.main(args=args)
  File "/home/dhood/ros2_ws/install_isolated/ros2topic/lib/python3.5/site-packages/ros2topic/verb/pub.py", line 43, in main
    return main(args)
  File "/home/dhood/ros2_ws/install_isolated/ros2topic/lib/python3.5/site-packages/ros2topic/verb/pub.py", line 47, in main
    publisher(args.message_type, args.topic_name, args.values)
  File "/home/dhood/ros2_ws/install_isolated/ros2topic/lib/python3.5/site-packages/ros2topic/verb/pub.py", line 69, in publisher
    setattr(msg, field_name, field_type(field_value))
TypeError: __init__() takes 1 positional argument but 2 were given
```

dirk-thomas, 2017-06-27:
Please see #24 for a proposed fix.

PR #24 (2017-06-27): fix population of recursive message fields
Fixes #22.

Ready for review.
closed 2017-06-27 by dirk-thomas