ros2/ros2cli #32: Traceback when runnning ros2 topic list on windows

tfoote, 2017-06-28:
This happened once when I was testing. I'd run the command before and after successfully.

Using rmw_connext_cpp

```
C:\dev\ros2>ros2 topic list
Traceback (most recent call last):
  File "C:\dev\ros2\install\bin\ros2-script.py", line 11, in <module>
    load_entry_point('ros2cli==0.0.0', 'console_scripts', 'ros2')()
  File "C:\dev\ros2\install\Lib\site-packages\ros2cli\cli.py", line 64, in main
    rc = extension.main(parser=parser, args=args)
  File "C:\dev\ros2\install\Lib\site-packages\ros2topic\command\topic.py", line 38, in main
    return extension.main(args=args)
  File "C:\dev\ros2\install\Lib\site-packages\ros2topic\verb\list.py", line 34, in main
    with NodeStrategy(args) as node:
  File "C:\dev\ros2\install\Lib\site-packages\ros2cli\node\strategy.py", line 33, in __enter__
    self.node.__enter__()
  File "C:\dev\ros2\install\Lib\site-packages\ros2cli\node\daemon.py", line 67, in __enter__
    methods = self._proxy.system.listMethods()
  File "C:\Python36\lib\xmlrpc\client.py", line 1112, in __call__
    return self.__send(self.__name, args)
  File "C:\Python36\lib\xmlrpc\client.py", line 1452, in __request
    verbose=self.__verbose
  File "C:\Python36\lib\xmlrpc\client.py", line 1154, in request
    return self.single_request(host, handler, request_body, verbose)
  File "C:\Python36\lib\xmlrpc\client.py", line 1166, in single_request
    http_conn = self.send_request(host, handler, request_body, verbose)
  File "C:\Python36\lib\xmlrpc\client.py", line 1279, in send_request
    self.send_content(connection, request_body)
  File "C:\Python36\lib\xmlrpc\client.py", line 1309, in send_content
    connection.endheaders(request_body)
  File "C:\Python36\lib\http\client.py", line 1234, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "C:\Python36\lib\http\client.py", line 1026, in _send_output
    self.send(msg)
  File "C:\Python36\lib\http\client.py", line 964, in send
    self.connect()
  File "C:\Python36\lib\http\client.py", line 936, in connect
    (self.host,self.port), self.timeout, self.source_address)
  File "C:\Python36\lib\socket.py", line 722, in create_connection
    raise err
  File "C:\Python36\lib\socket.py", line 713, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it
```

dirk-thomas, 2017-06-28:
If you can't reproduce the problem I would suggest to close the ticket. Since there is not much we can do about it if it is not reproducible.

tfoote, 2017-06-29:
I got the same error with a different invocation in my rmw_fastrtps window. Rerunning it immediately afterwards was fine.

```
C:\dev\ros2>ros2 node list
Traceback (most recent call last):
  File "C:\dev\ros2\install\bin\ros2-script.py", line 11, in <module>
    load_entry_point('ros2cli==0.0.0', 'console_scripts', 'ros2')()
  File "C:\dev\ros2\install\Lib\site-packages\ros2cli\cli.py", line 64, in main
    rc = extension.main(parser=parser, args=args)
  File "C:\dev\ros2\install\Lib\site-packages\ros2node\command\node.py", line 34, in main
    return extension.main(args=args)
  File "C:\dev\ros2\install\Lib\site-packages\ros2node\verb\list.py", line 34, in main
    with NodeStrategy(args) as node:
  File "C:\dev\ros2\install\Lib\site-packages\ros2cli\node\strategy.py", line 33, in __enter__
    self.node.__enter__()
  File "C:\dev\ros2\install\Lib\site-packages\ros2cli\node\daemon.py", line 67, in __enter__
    methods = self._proxy.system.listMethods()
  File "C:\Python36\lib\xmlrpc\client.py", line 1112, in __call__
    return self.__send(self.__name, args)
  File "C:\Python36\lib\xmlrpc\client.py", line 1452, in __request
    verbose=self.__verbose
  File "C:\Python36\lib\xmlrpc\client.py", line 1154, in request
    return self.single_request(host, handler, request_body, verbose)
  File "C:\Python36\lib\xmlrpc\client.py", line 1166, in single_request
    http_conn = self.send_request(host, handler, request_body, verbose)
  File "C:\Python36\lib\xmlrpc\client.py", line 1279, in send_request
    self.send_content(connection, request_body)
  File "C:\Python36\lib\xmlrpc\client.py", line 1309, in send_content
    connection.endheaders(request_body)
  File "C:\Python36\lib\http\client.py", line 1234, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "C:\Python36\lib\http\client.py", line 1026, in _send_output
    self.send(msg)
  File "C:\Python36\lib\http\client.py", line 964, in send
    self.connect()
  File "C:\Python36\lib\http\client.py", line 936, in connect
    (self.host,self.port), self.timeout, self.source_address)
  File "C:\Python36\lib\socket.py", line 722, in create_connection
    raise err
  File "C:\Python36\lib\socket.py", line 713, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it

C:\dev\ros2>ros2 node list

C:\dev\ros2>
```

tfoote, 2017-06-29:
service list too
```
C:\dev\ros2>ros2 service list
Traceback (most recent call last):
  File "C:\dev\ros2\install\bin\ros2-script.py", line 11, in <module>
    load_entry_point('ros2cli==0.0.0', 'console_scripts', 'ros2')()
  File "C:\dev\ros2\install\Lib\site-packages\ros2cli\cli.py", line 64, in main
    rc = extension.main(parser=parser, args=args)
  File "C:\dev\ros2\install\Lib\site-packages\ros2service\command\service.py", line 38, in main
    return extension.main(args=args)
  File "C:\dev\ros2\install\Lib\site-packages\ros2service\verb\list.py", line 34, in main
    with NodeStrategy(args) as node:
  File "C:\dev\ros2\install\Lib\site-packages\ros2cli\node\strategy.py", line 33, in __enter__
    self.node.__enter__()
  File "C:\dev\ros2\install\Lib\site-packages\ros2cli\node\daemon.py", line 67, in __enter__
    methods = self._proxy.system.listMethods()
  File "C:\Python36\lib\xmlrpc\client.py", line 1112, in __call__
    return self.__send(self.__name, args)
  File "C:\Python36\lib\xmlrpc\client.py", line 1452, in __request
    verbose=self.__verbose
  File "C:\Python36\lib\xmlrpc\client.py", line 1154, in request
    return self.single_request(host, handler, request_body, verbose)
  File "C:\Python36\lib\xmlrpc\client.py", line 1166, in single_request
    http_conn = self.send_request(host, handler, request_body, verbose)
  File "C:\Python36\lib\xmlrpc\client.py", line 1279, in send_request
    self.send_content(connection, request_body)
  File "C:\Python36\lib\xmlrpc\client.py", line 1309, in send_content
    connection.endheaders(request_body)
  File "C:\Python36\lib\http\client.py", line 1234, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "C:\Python36\lib\http\client.py", line 1026, in _send_output
    self.send(msg)
  File "C:\Python36\lib\http\client.py", line 964, in send
    self.connect()
  File "C:\Python36\lib\http\client.py", line 936, in connect
    (self.host,self.port), self.timeout, self.source_address)
  File "C:\Python36\lib\socket.py", line 722, in create_connection
    raise err
  File "C:\Python36\lib\socket.py", line 713, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it

C:\dev\ros2>
```

When this was launched I checked and the daemon was not in the process list. A 2nd run it successfully ran, and the daemon was appeared in the process list. It seems that something might have failed when launching the daemon that first time.

dirk-thomas, 2017-06-29:
I gave this another try and still can't reproduce the problem I added the "more-information-needed" label.

dirk-thomas, 2017-06-30:
@tfoote What is the status on this with the current state (since before you were using an outdated version)?

tfoote, 2017-06-30:
I haven't seen it, but I also haven't been running many things that ping the daemon. I'll close it for now. If anyone else sees this issue please comment back here. We can reopen if there's evidence of an ongoing issue.

closed 2017-06-30 by tfoote