ros2/ros2cli #34: daemon prevents rebuilding due to having dll open

tfoote, 2017-06-30:
After using the ros2 cli the daemon prevents a rebuild from completing successfully on windows. 

A way to have introspection as to the cause of this failure would be very helpful. Or else a workaround that would allow the build to request the daemon to shutdown or other technique to not block the build would provide a better user experience. 

```
  CMake Error at cmake_install.cmake:46 (file):
    file INSTALL cannot copy file "C:/dev/ros2/build/rcl/Release/rcl.dll" to
    "C:/dev/ros2/install/bin/rcl.dll".

C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: The command "
setlocal\r [C:\dev\ros2\build\rcl\INSTALL.vcxproj]
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: "C:\Program F
iles\CMake\bin\cmake.exe" -DBUILD_TYPE=Release -P cmake_install.cmake\r [C:\dev\ros2\build\rcl\INSTALL.vcxproj]
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: if %errorleve
l% neq 0 goto :cmEnd\r [C:\dev\ros2\build\rcl\INSTALL.vcxproj]
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: :cmEnd\r [C:\
dev\ros2\build\rcl\INSTALL.vcxproj]
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: endlocal & ca
ll :cmErrorLevel %errorlevel% & goto :cmDone\r [C:\dev\ros2\build\rcl\INSTALL.vcxproj]
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: :cmErrorLevel
\r [C:\dev\ros2\build\rcl\INSTALL.vcxproj]
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: exit /b %1\r
[C:\dev\ros2\build\rcl\INSTALL.vcxproj]
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: :cmDone\r [C:
\dev\ros2\build\rcl\INSTALL.vcxproj]
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: if %errorleve
l% neq 0 goto :VCEnd\r [C:\dev\ros2\build\rcl\INSTALL.vcxproj]
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: :VCEnd" exite
d with code 1. [C:\dev\ros2\build\rcl\INSTALL.vcxproj]
Done Building Project "C:\dev\ros2\build\rcl\INSTALL.vcxproj" (default targets) -- FAILED.

Build FAILED.

"C:\dev\ros2\build\rcl\INSTALL.vcxproj" (default target) (1) ->
(PostBuildEvent target) ->
  C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: The command
 "setlocal\r [C:\dev\ros2\build\rcl\INSTALL.vcxproj]
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: "C:\Program F
iles\CMake\bin\cmake.exe" -DBUILD_TYPE=Release -P cmake_install.cmake\r [C:\dev\ros2\build\rcl\INSTALL.vcxproj]
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: if %errorleve
l% neq 0 goto :cmEnd\r [C:\dev\ros2\build\rcl\INSTALL.vcxproj]
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: :cmEnd\r [C:\
dev\ros2\build\rcl\INSTALL.vcxproj]
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: endlocal & ca
ll :cmErrorLevel %errorlevel% & goto :cmDone\r [C:\dev\ros2\build\rcl\INSTALL.vcxproj]
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: :cmErrorLevel
\r [C:\dev\ros2\build\rcl\INSTALL.vcxproj]
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: exit /b %1\r
[C:\dev\ros2\build\rcl\INSTALL.vcxproj]
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: :cmDone\r [C:
\dev\ros2\build\rcl\INSTALL.vcxproj]
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: if %errorleve
l% neq 0 goto :VCEnd\r [C:\dev\ros2\build\rcl\INSTALL.vcxproj]
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: :VCEnd" exite
d with code 1. [C:\dev\ros2\build\rcl\INSTALL.vcxproj]

    0 Warning(s)
    1 Error(s)

Time Elapsed 00:00:04.53
```

dirk-thomas, 2017-06-30:
Since many packages contribute artifacts which are being used by the daemon I don't think it is feasible (or even desirable) to make their build logic aware of this. The same problem will happen if the user runs any executable while trying to build the package containing it at the same time.

Please feel free to propose any solution for this. I can only imagine a command line tool to manually ask for the current status of the daemon, and allow to manually start and stop it.

dirk-thomas, 2017-10-05:
I will close this ticket since I assume it is not fixable on a system which locks files when they are in use. Please feel free to comment on the closed ticket. If there is anything we can do I am happy to reopen the issue.

closed 2017-10-05 by dirk-thomas