ros2/ros2cli #37: ros2 service call does not wait for the service to become available

clalancette, 2017-06-30:
If I start out with nothing at all running, then run:

```
$ ros2 service call /add_two_ints example_interfaces/AddTwoInts
```

And then in a second terminal, I run:

```
$ ros2 run demo_nodes_cpp add_two_ints_server
```

The service request never comes through.  If I reverse the order (starting the server first), then it works fine.  It seems like `ros2 service call` is not waiting for the service to become available before sending the request.

dirk-thomas, 2017-06-30:
I think that is expected behavior. There is no wait-for-service functionality in Python at the moment. And even if there is I am not sure the command line invocation should wait for it (longer than a couple of seconds for discovery reasons).

clalancette, 2017-06-30:
Obviously the fact that wait_for_service() is not implemented in python is a blocker for doing this right now.  However, let's assume for a minute that we get it implemented in python.

It seems like most of the demos we have for clients do wait around for the service to become available, so it seems kind of incongruous that the command-line utility doesn't do the same thing.  For comparison, what does the equivalent ROS1 command do?

dirk-thomas, 2017-06-30:
It waits forever as far as I know.

Anyway I would argue that we shouldn't keep this ticket around since this is only a symptom of a feature we don't have. Otherwise we could fill multiple similar tickets which all come down to the same thing (since the same applies to all demos, tests, etc.). Therefore I would propose to close this for now.

mikaelarguedas, 2017-06-30:
:+1: this is captured by https://github.com/ros2/rclpy/issues/58 AFAIK

dirk-thomas, 2017-06-30:
~~It waits forever as far as I know.~~ **Update:** the command line tool will fail if the service isn't available:

```
$ rosservice call /foo std_srvs/Empty
ERROR: Service [/foo] is not available.
```

Anyway I would argue that we shouldn't keep this ticket around since this is only a symptom of a feature we don't have. Otherwise we could fill multiple similar tickets which all come down to the same thing (since the same applies to all demos, tests, etc.). Therefore I would propose to close this for now.

dirk-thomas, 2017-06-30:
Since this topic is tracker in `rclpy` I will close this ticket for now.

closed 2017-06-30 by dirk-thomas