ros2/ros2cli #40: Run arguments do not get passed to node executable

bponsler, 2017-07-11:
When attempting to pass command line arguments to a node using the **ros2 run [pkg] [exec]** syntax, the arguments get processed by the ros2 executable rather than the node executable being launched.

Example:

```
$ ros2 run ros1_bridge dynamic_bridge --help
usage: ros2 run [-h] package_name executable_name [argv [argv ...]]

Run a package specific executable

positional arguments:
  package_name     Name of the ROS package
  executable_name  Name of the executable
  argv             Pass arbitrary arguments to the executable (use '--' before
                   these arguments to ensure they are not handle by this
                   command)

optional arguments:
  -h, --help       show this help message and exit
```

Expected output:

```
$ ./install/lib/ros1_bridge/dynamic_bridge --help
Usage:
 -h, --help: This message.
 --show-introspection: Print output of introspection of both sides of the bridge.
 --print-pairs: Print a list of the supported ROS 2 <=> ROS 1 conversion pairs.
 --bridge-all-topics: Bridge all topics in both directions, whether or not there is a matching subscriber.
 --bridge-all-1to2-topics: Bridge all ROS 1 topics to ROS 2, whether or not there is a matching subscriber.
 --bridge-all-2to1-topics: Bridge all ROS 2 topics to ROS 1, whether or not there is a matching subscriber.
```

Although I'm sure you get the picture, here's another example:

```
$ ros2 run ros1_bridge dynamic_bridge --bridge-all-topics
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: unrecognized arguments: --bridge-all-topics
```

Add one passing just a flag:

```
$ ros2 run ros1_bridge dynamic_bridge -f
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: unrecognized arguments: -f
```

Arguments that are not prefixed with one or more dashes do allow the node to get executed (although I don't have evidence at the moment that they successfully get passed to the node).

E.g., (I don't have a master running, but it runs the bridge)

```
$ ros2 run ros1_bridge dynamic_bridge this_is_an_argument another arg3
[ERROR] [1499788917.336544285] [/ros_bridge]: [registerPublisher] Failed to contact master at [localhost:11311].  Retrying...
```

wjwwood, 2017-07-11:
From the `--help`:

```
argv             Pass arbitrary arguments to the executable (use '--' before
                   these arguments to ensure they are not handle by this
                   command)
```

You have to run it like `ros2 run ros1_bridge dynamic_bridge -- -f`.

I agree that this is a) not obvious and b) inconvenient, but it was the easiest thing to do without reimplementing the argument parsing rather than use Python's `argparse`.

We can either leave this open until the usability is improved or we can close this one and you may open another with that feature request.

dirk-thomas, 2017-07-11:
Please try what is recommended in the help text of `ros2`, specifically:

```
  argv             Pass arbitrary arguments to the executable (use '--' before
                   these arguments to ensure they are not handle by this
                   command)
```

Duplicate of #28 and #16.

dirk-thomas, 2017-07-11:
> We can either leave this open until the usability is improved or we can close this one and you may open another with that feature request.

I would prefer not to try to workaround Pythons way of argument parsing. This is a common way of doing it (e.g. ``gdb --args`). Therefore I would suggest to rather close this.

wjwwood, 2017-07-11:
> I would prefer not to try to workaround Pythons way of argument parsing. This is a common way of doing it (e.g. ``gdb --args`). Therefore I would suggest to rather close this.

I disagree, the underlying technology shouldn't dictate how the tool works. Especially since three different people have run into this and found it confusing enough to report it as an issue. I think we need to either do a better job detecting and reporting the issue or make it so that the `--` is not required.

bponsler, 2017-07-11:
Apologies for not doing a good enough job searching through the issues...or reading the help...

wjwwood, 2017-07-11:
> Apologies for not doing a good enough job searching through the issues...or reading the help...

No worries. As I pointed out, if you're the third person to run into this or miss the help, then we need to do something different so it's more obvious or so that it works as people expect it should, avoiding the surprise.

dhood, 2017-07-11:
> if you're the third person to run into this or miss the help, then we need to do something different so it's more obvious or so that it works as people expect it should, avoiding the surprise.

+1. I added a quick mention to the wiki for now (it's linked to from release notes). https://github.com/ros2/ros2/wiki/Introspection-with-command-line-tools

dirk-thomas, 2017-12-03:
Addressed by #61.

closed 2017-12-03 by dirk-thomas