ros2/ros2cli #51: r2b3 ros2param crashes with `list /`

sloretz, 2017-09-13:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10,
- Installation type:
  - http://ci.ros2.org/view/packaging/job/packaging_windows/728/artifact/ws/ros2-package-windows-AMD64.zip
- Version or commit hash:
  - def304b4270b6b935b7d6fbbf7671634c2532452
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  -  N/A

#### Steps to reproduce issue

```
ros2 run demo_nodes_cpp ros2param -- list /
```

#### Expected behavior
Not sure.

#### Actual behavior
ros2param.exe has stopped working

[image]

#### Additional information
Not from source, so unable to get a meaningful stack trace.

wjwwood, 2017-09-13:
I don't get this on my from source build, using master, pulled just about an hour ago.

[image]

wjwwood, 2017-09-13:
Were you running anything else at the time?

dirk-thomas, 2017-09-13:
Please move the ticket to the repo containing the failing script: https://github.com/ros2/demos/blob/6363be2efe2fea799d92bc22a66e776b2ca9c5d0/demo_nodes_cpp/src/parameters/ros2param.cpp

Also not that the C++ implementation of this command line tool is kind of EOL and is supposed to be replaced by a Python tool within the `ros2cli` repo. So I would suggest not to spend time on it.

mikaelarguedas, 2017-09-13:
hmm looks like it happens with "/" I tried random other names and if the node doesnt exist I get `service not available, waiting again...`

> Were you running anything else at the time?

I can reproduce the error above regardless of if another node with parameter server is running or not

While I agree the crash is bad, this I don't really see any use case for an empty node name (resulting in a FQN "/")

sloretz, 2017-09-13:
Oops, sorry about the wrong repo again. Moving

sloretz, 2017-09-13:
This issue was moved to ros2/demos#174

closed 2017-09-13 by sloretz