ros2/ros2cli #52: ros2 pkg executables list scripts as executables on windows

sloretz, 2017-09-13:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - http://ci.ros2.org/view/packaging/job/packaging_windows/732/artifact/ws/ros2-package-windows-AMD64.zip
- Version or commit hash:
  - def304b4270b6b935b7d6fbbf7671634c2532452
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
```
ros2 pkg executables examples_rclpy_minimal_publisher
```

#### Expected behavior
I'd expect only executable things to appear in the list
```
C:\Users\osrf>ros2 pkg executables examples_rclpy_minimal_publisher
examples_rclpy_minimal_publisher publisher_local_function.exe
examples_rclpy_minimal_publisher publisher_member_function.exe
examples_rclpy_minimal_publisher publisher_old_school.exe
```

#### Actual behavior
Both the `*.exe` and `*-script.py` appear in the list, but only one of them is executable
```
C:\Users\osrf>ros2 pkg executables examples_rclpy_minimal_publisher
examples_rclpy_minimal_publisher publisher_local_function-script.py
examples_rclpy_minimal_publisher publisher_local_function.exe
examples_rclpy_minimal_publisher publisher_member_function-script.py
examples_rclpy_minimal_publisher publisher_member_function.exe
examples_rclpy_minimal_publisher publisher_old_school-script.py
examples_rclpy_minimal_publisher publisher_old_school.exe
```

The script cannot be executed using ros2 run, so it probably shouldn't be in the list.
```
C:\Users\osrf>ros2 run examples_rclpy_minimal_publisher publisher_local_function-script.py
Traceback (most recent call last):
  File "C:\dev\ros2\Scripts\ros2-script.py", line 11, in <module>
    load_entry_point('ros2cli==0.0.0', 'console_scripts', 'ros2')()
  File "C:\dev\ros2\Lib\site-packages\ros2cli\cli.py", line 64, in main
    rc = extension.main(parser=parser, args=args)
  File "C:\dev\ros2\Lib\site-packages\ros2run\command\run.py", line 73, in main
    return run_executable(path=path, argv=args.argv, prefix=prefix)
  File "C:\dev\ros2\Lib\site-packages\ros2run\api\__init__.py", line 56, in run_executable
    process = subprocess.Popen(cmd)
  File "C:\Python36\lib\subprocess.py", line 707, in __init__
    restore_signals, start_new_session)
  File "C:\Python36\lib\subprocess.py", line 992, in _execute_child
    startupinfo)
OSError: [WinError 193] %1 is not a valid Win32 application
```

#### Additional information

dirk-thomas, 2017-09-13:
The script only shows files which have the executable bit set (https://github.com/ros2/ros2cli/blob/b962b78e160342aea5cfcfb9dd11edaf95af80cc/ros2pkg/ros2pkg/api/__init__.py#L53). And these files are executable. Invoking `<prefix>\Lib\examples_rclpy_minimal_publisher\publisher_local_function-script.py` works just fine.

So the questions are:
* why does `ros2 run` fails to invoke the script?
* how could `ros2 pkg executables` distinguish those in order to hide them?

dirk-thomas, 2017-09-13:
One option would be to prepend `sys.executable` to the subprocess invocation if the executable ends in `.py` on Windows.

dirk-thomas, 2017-09-13:
@sloretz Can you please double check that #53 works for you too.

PR #53 (2017-09-13): invoke Python scripts on Windows with interpreter
Fixes #52.
closed 2017-09-13 by dirk-thomas