ros2/ros2cli #60: ros2 topic list needs an option for pub/sub counts

BrannonKing, 2017-10-25:
Please add an option to show the number of publishers and subscribers on each topic as part listing them.

mikaelarguedas, 2017-10-25:
That would be a useful addition indeed.
Not sure if this should be part of the "list" verb or if a verb "info" like in ROS 1 would be a better place

dirk-thomas, 2017-10-25:
Isn't [--count-topics](https://github.com/ros2/ros2cli/blob/2409d8b230744608c40daf72189ac3e894435b60/ros2topic/ros2topic/verb/list.py#L30) providing  this already?

mikaelarguedas, 2017-10-25:
`count topics` gives the total number of topics but no insight on the publishers/subscribers associated with them. As I understand the request, @BrannonKing would like to have the number of pub sub associated / matched for each topic

dirk-thomas, 2017-10-25:
Thanks for clarifying. Yeah, that sounds like `ros2 topic info` (which doesn't exist at the moment).

Please consider to contribute a pull request to add this feature. That will likely speed up the process significantly.

BrannonKing, 2017-10-25:
While I agree that it should show in "info" as well (aka, on a particular topic), it's also very useful on the overall topic list. In "info" I want to see more than the count -- I want to see the actual node names of publishers and subscribers.

mikaelarguedas, 2017-10-26:
It looks like this information should come from the (currently not implemented) `ros2 topic info`. `ros2 topic info` will give details about all the publishers and subscribers related to a topic. `ros2 topic list` could then provide an optional argument that would leverage the results of `ros2 topic info` and display only the count of pub/sub for each topic

aarush22, 2017-10-29:
I would like to help with this issue if no one is working on it yet. As I understand A new verb '''info.py''' needs to be created in the '''ros2 topic verb''', could you guide me a little, how to get the pub/sub info for a topic.

Karsten1987, 2017-11-07:
Thanks @aarush22 for willing to help and please apologize for not getting feedback earlier. Any help is appreciated.

A few pointers to get you started:

In order to integrate a new verb into the ros2cli, you can basically just look at the existing verbs and apply your new verb accordingly. There is currently one open PR, integrating a new verb, which could serve you as a reference: https://github.com/ros2/ros2cli/pull/42

Then, in order to fetch information about topics, i.e. get the size of matched subscriptions and publishers, there are currently two functions in [rclcpp](https://github.com/ros2/rclcpp/blob/master/rclcpp/src/rclcpp/node_interfaces/node_graph.cpp#L166-L208) which get you the number of matched subscriptions and publishers.

Unfortunately, they are currently not implemented in [rclpy](https://github.com/ros2/rclpy), however, I don't believe it's too hard to expose this in python as well. You can as well look at the current [implementation of rclpy](https://github.com/ros2/rclpy/blob/master/rclpy/src/rclpy/_rclpy.c) to see how we currently access the rcl or rmw layer through python.

Nickolaim, 2018-03-27:
Hey, is this issue still actual?  I will be happy to work on this.

Karsten1987, 2018-03-27:
@aarush22 Is there any update from your side? Could you have a chance to work on this already?

@Nickolaim, thanks! We appreciate the help. I am not aware of any open PR tackling this issue. If there is no ongoing work, I believe you can feel free to start looking at this problem. I would recommend though that you report early on this issue to avoid possible conflicts or duplicated work.

Nickolaim, 2018-03-28:
Feel free to assign it to me.  Not sure it will be done in 2 weeks, but I think I understand what needs to be done.

Karsten1987, 2018-03-28:
Sounds great. Thanks a lot! I marked this issue as "in progress" to keep track of it in our weekly review. 
Needless to say, feel free to comment on this ticket in case you need further help or assist.

aarush22, 2018-04-01:
@Karsten1987  sorry for the late reply, but I've been abroad for work related stuff since a few months, and I've not been able to do significant work on this issue yet, so if you want to work on this issue @Nickolaim , no conflicts from my side.

Nickolaim, 2018-04-02:
@Karsten1987 - I am following the code, and it looks like that [class Node in python](https://github.com/ros2/rclpy/blob/master/rclpy/rclpy/node.py#L52) needs to implement `count_publishers()` and `count_subscibers()` that are going to call C implementation in `rclpy/_rclpy.c`, and this methods need to call [methods from NodeGraph in rclcpp](https://github.com/ros2/rclcpp/blob/master/rclcpp/src/rclcpp/node_interfaces/node_graph.cpp#L166-L208).

But Node class in python already has publishers and subscribers.  If I just add `count_publishers()` and `count_subscibers()`, that is ignoring members of the class it will be super confusing.

Another implementation might have publishers and subscribers being initialized properly in Node constructor.  This potentially raises an issue of ownership and cleanup (should the Node when closed cleanup publishers and subscribers it does not own?).  But this implementation will allow having more informative `topic info`, that can print all publishers and subscribers, not only the count.

What do you think?

Nickolaim, 2018-04-05:
It took me a while to understand that I don't need to call the code from `rclcpp`, I just need to do the same implementation in `rclpy`. [The first version is in the link](https://github.com/Nickolaim/rclpy/commit/d66aff2bdf1b941deabe2135b5b79f25880f3fb6).

I don't like the fact that 2 new functions in `_rclpy.c` have 90% of the same code; most likely will refactor the actual work into a separate function that accepts `rcl_count_*` function pointer and either `"subscriber"` or `"publisher"`.  Also `rclcpp` calls `expand_topic_or_service_name()` I guess I need to do the same.

My biggest problem is how to write a meaningful unit test.

Also, there are a bunch of warnings from formatting tools and they are outside of my change.  Is it expected?

dirk-thomas, 2018-04-05:
To clarify the two relevant but different information:

1. having a node one can query the number of publishers and subscribers for a given topic name (we have some API in place for that)

2. for a given publisher or subscriber one can query the number of *matched* subscribers / publishers (we don't have that functionality, not even in the rmw interface)

Consider the following scenario (all using the same topic name):
* one publisher uses reliable
* one publisher uses best-effort
* one subscriber demands reliable

Clearly 1. will return 2 publishers and 1 subscriber. But 2. should return:
* for the first publisher that it has 1 subscriber
* for the second publisher that it has no subscribers
* for the subscriber that it has only 1 publisher

Nickolaim, 2018-04-05:
I’d like to limit the scope for now for the scenario #1, that is going to return 2 publishers and 1 subscriber.

The next version should probably list names of publishers and subscribers.

Karsten1987, 2018-04-09:
I've just tested your PR locally and it seems to work fine from what I can tell.
However, you might want to run tests on your PR. I've noticed that some style issues came up when running the tests locally on my machine (i.e. flake8 for ros2topic and uncrustify for rclpy).

mikaelarguedas, 2018-04-09:
The list of dependencies we use for testing can be found on the ["building from source" instructions](https://github.com/ros2/ros2/wiki/Installation#building-from-source) (e.g. [for Linux](https://github.com/ros2/ros2/wiki/Linux-Development-Setup#how-to-setup-the-development-environment)). As some of them are pip packages you may need to upgrade them to the latest version.

Nickolaim, 2018-04-10:
Issues from static code analysis tools are fixed in both PRs.

I want to set up Travis CI for ros2cli (at least).  Any reason why I should not be doing it?

dirk-thomas, 2018-04-10:
> I want to set up Travis CI for ros2cli (at least). Any reason why I should not be doing it?

In the future these packages should test actual functionality and not only likes. A simple Travis job not building all of ROS 2 below it will not be sufficient to perform all the tests. Therefore I don't think setting up Travis makes sense.

Karsten1987, 2018-04-10:
CI:

linux: [image]
osx: [image]
win: [image]

It looks good to me. @dirk-thomas, any thoughts?
I would merge it as an incremental improvement.

dirk-thomas, 2018-04-12:
@Karsten1987 I commented on one of the PRs with required changes. Please review the PRs explicitly beside the comment and CI in this meta ticket since otherwise that is not visible when looking at the PRs.