ros2/ros2cli #62: Double free killing service call to nonexistent server

dhood, 2017-11-16:
Reproduce with: `ros2 service call /add_two_ints example_interfaces/AddTwoInts` when no server is around, then ctrl-c it after it has sent the request. I get a core dump like in https://gist.github.com/dhood/ea5efa71ad2b3fa47a8942058f2de82a:
```
$ ros2 service call /add_two_ints example_interfaces/AddTwoInts
requester: making request: example_interfaces.srv.AddTwoInts_Request(a=0, b=0)

^C*** Error in `/usr/bin/python3': double free or corruption (fasttop): 0x00000000031b7530 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7ff74a28d7e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7ff74a29637a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7ff74a29a53c]
/home/dhood/ros2_ws/install_isolated/rclpy/lib/python3.5/site-packages/rclpy/_rclpy.cpython-35m-x86_64-linux-gnu.so(+0x8129)[0x7ff7477a2129]
/usr/bin/python3(PyCFunction_Call+0x4f)[0x4e9b9f]
/usr/bin/python3(PyEval_EvalFrameEx+0x614)[0x524414]
/usr/bin/python3(PyEval_EvalFrameEx+0x4a14)[0x528814]
/usr/bin/python3(PyEval_EvalCodeEx+0x13b)[0x52e12b]
/usr/bin/python3[0x4ebcc3]
/usr/bin/python3(PyObject_Call+0x47)[0x5b7167]
...
```

This does not occur with `ros2 run demo_nodes_py add_two_ints_client` but I still haven't found what the significant difference is. I even tried pasting the `add_two_ints_client` main in place of https://github.com/ros2/ros2cli/blob/cf17fdabbb62015418ca76eaabfa141e0826da08/ros2service/ros2service/verb/call.py#L57 but it still gives the double free error. `add_two_ints_client`, on the other hand, gives:
```
$ ros2 run demo_nodes_py add_two_ints_client
^CTraceback (most recent call last):
  File "/home/dhood/ros2_ws/install_isolated/demo_nodes_py/lib/demo_nodes_py/add_two_ints_client", line 11, in <module>
    load_entry_point('demo-nodes-py==0.0.0', 'console_scripts', 'add_two_ints_client')()
  File "/home/dhood/ros2_ws/install_isolated/demo_nodes_py/lib/python3.5/site-packages/demo_nodes_py/services/add_two_ints_client.py", line 39, in main
    node.destroy_node()
  File "/home/dhood/ros2_ws/install_isolated/rclpy/lib/python3.5/site-packages/rclpy/node.py", line 266, in destroy_node
    _rclpy.rclpy_destroy_node_entity(cli.client_handle, self.handle)
RuntimeError: Failed to fini 'rcl_client_t': rcl node is invalid, rcl instance id does not match, at /home/dhood/ros2_ws/src/ros2/rcl/rcl/src/rcl/node.c:374
Exception ignored in: <bound method Node.__del__ of <rclpy.node.Node object at 0x7fb28263ce80>>
Traceback (most recent call last):
  File "/home/dhood/ros2_ws/install_isolated/rclpy/lib/python3.5/site-packages/rclpy/node.py", line 295, in __del__
    self.destroy_node()
  File "/home/dhood/ros2_ws/install_isolated/rclpy/lib/python3.5/site-packages/rclpy/node.py", line 266, in destroy_node
    _rclpy.rclpy_destroy_node_entity(cli.client_handle, self.handle)
RuntimeError: Failed to fini 'rcl_client_t': rcl node is invalid, rcl instance id does not match, at /home/dhood/ros2_ws/src/ros2/rcl/rcl/src/rcl/node.c:374
```

Perhaps `ros2 service call` doesn't manage to catch the exceptions which is how it gets into trouble.

dirk-thomas, 2017-11-17:
When being aborted the script tries to call `destroy_node` on the node. Within [that function](https://github.com/ros2/rclpy/blob/5c1b6e10b75166f14cd381a792bc4a9ffd429df0/rclpy/rclpy/node.py#L266) the it raises an exception (`PyCapsule_GetPointer called with invalid PyCapsule object`) when trying to destroy the client. The exception will basically abort the `destroy_node` function prematurely wihch didn't clean up all the other entities and also didn't reset `self._handle` to `None`. So while the exceptions is being handles the GC kicks in which tries to delete the not referenced `Node` instance. But the `__del__ method invokes `destroy_node` again which results in the double free.

Conclusion: there are actually two bugs in `rclpy`:

* the `destroy_node` function must make sure to leave the node instance in a "consistent" state when raising an exception
* another problem is that the destroy call of the client actually fails with an invalid PyCapsule, so somewhere that capsule is being wrongly invalidated

sloretz, 2017-11-17:
+1 to keeping the node state consistent during `destroy_node`. It should workaround the double free.

```
RuntimeError: Failed to fini 'rcl_client_t': rcl node is invalid, rcl instance id does not match, at /home/dhood/ros2_ws/src/ros2/rcl/rcl/src/rcl/node.c:374
```

I doubt this is an invalid PyCapsule. The first exception in `destroy_node` would have caused it to exit before destroying the node's handle. If it was a PyCapsule of the wrong type then `PyCapsule_GetPointer` would have raised `ValueError` in `rcl_client_fini` instead. I checked for `PyCapsule_New` being called with the string `rcl_node_t` and only found one in `rclpy_create_node` so I think it had to have been given a valid node pointer.

It could have been caused by `rclpy.shutdown()` being called before `destroy_node` because [shutdown resets the instance id to zero](https://github.com/ros2/rcl/blob/master/rcl/src/rcl/rcl.c#L51). If that's what happened here maybe raising an exception in `rclpy_destroy_node_entity()` is the wrong behavior? The memory used by `rcl_client_t` could be silently freed instead. However, it looks like the space used for `rcl_client_impl_t` is leaked anyways because `rcl_client_fini`will [return without deallocating it](https://github.com/ros2/rcl/blob/master/rcl/src/rcl/client.c#L176) when the node handle is invalid.

PR #65 (2017-12-03): wait for service before calling it
Connect to #62. Fixes #62.
closed 2017-12-03 by dirk-thomas