ros2/ros2cli #69: daemon behavior using multiple rmw implementations inconsistent

tfoote, 2017-12-07:
I've been testing fastrtps and opensplice in parallel environments on the same machine.

Testing (fastrtps) `ros2 node list` I noticed that the list had a lot of extra padding.
```
C:\Users\OSRF\Downloads\ardent_fastrtps\ros2-windows>ros2 node list
lc_listener

C:\Users\OSRF\Downloads\ardent_fastrtps\ros2-windows>
```
vs 
```
C:\Users\OSRF\Downloads\ardent_fastrtps\ros2-windows>ros2 node list
lc_listener

C:\Users\OSRF\Downloads\ardent_fastrtps\ros2-windows>ros2 daemon stop
```

When I restarted things again it went away. But after some more testing it came back. What I found was that the extra newlines come in when the daemon is the opensplice implementation's version. 

It took me a while to figure this out as `daemon status` will report running if either implementations daemon is running.

But `daemon stop` will only stop the environment's version of the daemon, and refuse to stop the other one. 
```
C:\Users\OSRF\Downloads\ardent_fastrtps\ros2-windows>ros2 daemon stop
Failed to shutdown daemon, it might be using a different rmw implementation
```

I'm not sure on the design goal of whether the daemon's are supposed to be implementation agnostic and ticket the different behavior between them. Or whether we should instead work to silo them so that we don't get cross talking behavior and each implementation runs it's own daemon in parallel.

dirk-thomas, 2017-12-07:
> Testing (fastrtps) `ros2 node list` I noticed that the list had a lot of extra padding.
> When I restarted things again it went away. But after some more testing it came back.

Please provide a reproducible example for this.

dhood, 2017-12-07:
ros2 node list padding spun off into https://github.com/ros2/ros2cli/issues/72. @dirk-thomas I would think you're still probably the best person to answer @tfoote's questions about the main questions about the daemon usage

dirk-thomas, 2017-12-07:
> I'm not sure on the design goal of whether the daemon's are supposed to be implementation agnostic and ticket the different behavior between them. Or whether we should instead work to silo them so that we don't get cross talking behavior and each implementation runs it's own daemon in parallel.

Currently we only spawn a single daemon since its API is agnostic to the rmw impl being used. The expectation is that they would return the information independent of their rmw impl. Therefore I don't think we should spawn separate daemons.

tfoote, 2017-12-07:
If we're only going to have one daemon and it should be independent of the implementation we should allow `ros2 daemon stop` to stop the other implementations daemon. Then I think everything will be symmetric and you won't care which daemon is running in the background.

dirk-thomas, 2017-12-07:
The internal rpc url currently has the rmw implementation which the command line tool can't guess if it is not matching. I don't recall why it was added and if it can easily be removed.

dirk-thomas, 2017-12-07:
The internal rpc url currently has the rmw implementation which the command line tool can't guess if it is not matching. I don't recall why it was added and if it can easily be removed.

Update: looking at #11 clarifies that this is intentional to avoid cross vendor answers from the daemon.

dirk-thomas, 2017-12-07:
The internal rpc url currently has the rmw implementation which the command line tool can't guess if it is not matching. I don't recall why it was added and if it can easily be removed.

Update: looking at #11 clarifies that this is intentional to avoid cross vendor answers from the daemon. And you can't have two daemons answering on the same port and we don't have a way to determine a deterministic port number specific to the rmw implementation so we can't run separate daemons for each rmw that easily.

dirk-thomas, 2018-02-22:
As mentioned before the inconsistent output is addressed by #76.

#80 changes the daemon to use a rmw impl. agnostic URL allowing them to "cross-talk".

Assuming that covers all items discussed in this issue the ticket will be closed by #80.

PR #80 (2018-02-27): use rmw agnostic daemon URL
Removes the rmw impl. specific path from the daemon URL. This allows a client with any rmw impl. to talk to a daemon with any rmw impl.

Fixes #69.
closed 2018-02-27 by dirk-thomas