ros2/ros2cli #72: Blank lines / "(no name)" in output of ros2 node list

dhood, 2017-12-07:
Not a major issue but something we should look into eventually:

built from source with fastrtps:

```
<start a server node>
dhood@osrf-esteve:~/ros2_ardent [ros2_ardent]$ ros2 node list
add_two_ints_server
<kill server node>
dhood@osrf-esteve:~/ros2_ardent [ros2_ardent]$ ros2 node list

<start it again>
dhood@osrf-esteve:~/ros2_ardent [ros2_ardent]$ ros2 node list

add_two_ints_server
dhood@osrf-esteve:~/ros2_ardent [ros2_ardent]$ ros2 node list

<kill it again>
dhood@osrf-esteve:~/ros2_ardent [ros2_ardent]$ 

```

built from source with connext on osx:
```
$ RMW_IMPLEMENTATION=rmw_connext_cpp ros2 node list
REDAWorker_enterExclusiveArea:worker U00007fff7a779300 deadlock risk: cannot enter 0x7fa690f9c770 of level 40 from level 40
REDACursor_modifyReadWriteArea:!enter worker (U00007fff7a779300)'s exclusive area
PRESPsService_removeRemoteEndpoint:!modify pres psRemoteWriter
REDAWorker_enterExclusiveArea:worker U00007fff7a779300 deadlock risk: cannot enter 0x7fa690f9c770 of level 40 from level 40
REDACursor_modifyReadWriteArea:!enter worker (U00007fff7a779300)'s exclusive area
PRESPsService_removeRemoteEndpoint:!modify pres psRemoteReader
add_two_ints_server
(no name)
(no name)
(no name)
(no name)
```

the `(no name)` output comes from this line in rmw_connext_cpp:
https://github.com/ros2/rmw_connext/blob/efaf513a5f9303216670ed76b5060ed5b8f958e5/rmw_connext_shared_cpp/src/node_names.cpp#L76

Perhaps this equivalent in fastrtps can return empty names: https://github.com/ros2/rmw_fastrtps/blob/3e5edb3316e2aee325e9354549e54c999a89bca9/rmw_fastrtps_cpp/src/rmw_node_names.cpp#L56

I haven't seen opensplice output empty lines.

Karsten1987, 2017-12-07:
I have the same output for fastrtps!

mikaelarguedas, 2017-12-07:
oh weird. Just to confirm: that's something you both see only on MacOS (and not on Linux)?

dhood, 2017-12-07:
I edited my first post to clarify the first output is from fastrtps on linux. @tfoote saw it on windows in https://github.com/ros2/ros2cli/issues/69 so this is cross-platform FWICT

dirk-thomas, 2017-12-07:
I ran the same sequence on Linux (build from source) and the node list is always correct for me (no sign of empty lines).

tfoote, 2017-12-07:
I have three ways to reproduce it.

# One

- FastRTPS window: `ros2 daemon start`
- OpenSplice window: `ros2 node list` OK with config file error
```
C:\Users\OSRF\Downloads\ardent_opensplice\ros2-windows>ros2 node list
========================================================================================
Report      : ERROR
Date        : Thu Dec 07 12:23:08 Pacific Standard Time 2017
Description : Failed to open configuration syntax file: C:/dev/opensplice/HDE/x86_64.win64\etc\metaconfig.xml
Node        : DESKTOP-1VF050Q
Process     : python.exe <5716>
Thread      : spliced 5688
Internals   : 6.7.171127OSS/fcc005a2/fcc005a2/configuration validator/cfg_validator.c/825/0/1512678188.615544300/0

C:\Users\OSRF\Downloads\ardent_opensplice\ros2-windows>
```
- FastRTPS window: `ros2 node list`
```
C:\Users\OSRF\Downloads\ardent_fastrtps\ros2-windows>ros2 node list

C:\Users\OSRF\Downloads\ardent_fastrtps\ros2-windows>
```
# two
- OpenSplice window `ros2 daemon start` 
- FastRTPS window: `ros2 node list`

```
C:\Users\OSRF\Downloads\ardent_fastrtps\ros2-windows>ros2 node list

C:\Users\OSRF\Downloads\ardent_fastrtps\ros2-windows>
```
# three

- FastRTPS window: `ros2 daemon start`
- OpenSplice window: `ros2 run lifecycle lifecycle_listener`
- FastRTPS window: `ros2 node list`
```
C:\Users\OSRF\Downloads\ardent_fastrtps\ros2-windows>ros2 node list
lc_listener

C:\Users\OSRF\Downloads\ardent_fastrtps\ros2-windows>
```

dhood, 2017-12-07:
I think having opensplice nodes running is what causes the extra output.

Having _only_ fastrtps/connext nodes I see expected output from `RMW_IMPLEMENTATION=rmw_fastrtps_cpp ros2 node list` (no padding) and `RMW_IMPLEMENTATION=rmw_connext_cpp ros2 node list` (no "(no name)").

When opensplice nodes are introduced the output happens.

I've noticed that `RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 node list`:
- is able to detect opensplice nodes without error
- does not report opensplice nodes as having shut down immediately
- cannot detect connext/fastrtps nodes

I suspect there may be general discovery incompatibility that causes `rmw_get_node_names` to have errors when fastrtps/connext are requesting status of opensplice participants, and it looks like opensplice can't detect the others either

dirk-thomas, 2017-12-07:
I don't think that this is related to the daemon at all.

You can try the following to reproduce:
* make sure no other processes are running
* start any node using OpenSplice (without `ros2 run` but invoking the executable directly)
* call `ros2 node list` using FastRTPS (which will start a daemon using FastRTPS)

which leads to the same result: the node list containing empty lines.

dirk-thomas, 2017-12-07:
The reason for the problem as well as a possible solution to fix this is described in ros2/ros2#438.

dirk-thomas, 2018-02-22:
Addressed by #76.

closed 2018-03-30 by dirk-thomas