ros2/ros2cli #74: Calling ros2 cli tool without arguments throws an error instead of listing available commands

jbohren, 2017-12-19:
Calling `ros2` cli tool without arguments throws an error instead of listing available commands.

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - binary
- Version or commit hash:
  - Ardent

#### Steps to reproduce issue
```
$ ros2
```

#### Expected behavior

```
$ ros2
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...
<<<LIST OF COMMANDS>>>
```

#### Actual behavior

```
$ ros2
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: the following arguments are required: Call `ros2 <command> -h` for more detailed usage.
```

#### Bug Source

It looks like this is being passed as the `metavar` instead of the help text to the argument parser.
https://github.com/ros2/ros2cli/blob/a4e00b2e3846db9e7b64431110f81d02f6efa0ee/ros2cli/ros2cli/command/__init__.py#L92-L95

dirk-thomas, 2017-12-19:
Even without the `metavar` parameter the default behavior of `argparse` will print something like this:

```
usage: ros2 [-h]
            {daemon,extension_points,extensions,msg,node,pkg,run,security,service,srv,topic}
            ...
ros2: error: the following arguments are required:  command
```

Maybe the full help text should be shown in this case (same as when calling `ros2 -h`) that will require to workaround the default behavior of `argparse` though. Any pull requests to improve this would be very welcome.

dirk-thomas, 2018-02-22:
Addressed by #81.

PR #81 (2018-02-22): print full help when no command is passed
When no `command` / `verb` is passed the full help is printed as if the user would have passed `--help`.

Fixes #74.

The API is backward compatible so that the behavior of existing extensions isn't changes. Extensions implemented in other repositories could be updated in the same way (not to require the `verb`) if they want to show the full help text in that case.
closed 2018-02-22 by dirk-thomas