ros2/ros2cli #86: ros2 pkg create: resource files not installed

dhood, 2018-04-02:
How have people been testing `ros2 pkg create`? I tried to run it but the templates aren't installed so I get:

```
$ ros2 pkg create dummy_package --license asdf
going to create a new package
package name: dummy_package
destination directory: /tmp
package format: 2
version: 0.0.0
description: TODO: Package description
maintainer: ['dhood <dhood@users.noreply.github.com>']
licenses: ['asdf']
build type: ament_cmake
dependencies: []
creating folder ./dummy_package
Traceback (most recent call last):
  File "/home/dhood/ros2_ws/install_isolated/ros2cli/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.4.0', 'console_scripts', 'ros2')()
  File "/home/dhood/ros2_ws/install_isolated/ros2cli/lib/python3.5/site-packages/ros2cli/cli.py", line 69, in main
    rc = extension.main(parser=parser, args=args)
  File "/home/dhood/ros2_ws/install_isolated/ros2pkg/lib/python3.5/site-packages/ros2pkg/command/pkg.py", line 39, in main
    return extension.main(args=args)
  File "/home/dhood/ros2_ws/install_isolated/ros2pkg/lib/python3.5/site-packages/ros2pkg/verb/create.py", line 154, in main
    create_package_environment(package, args.destination_directory)
  File "/home/dhood/ros2_ws/install_isolated/ros2pkg/lib/python3.5/site-packages/ros2pkg/api/create.py", line 97, in create_package_environment
    package_xml_config)
  File "/home/dhood/ros2_ws/install_isolated/ros2pkg/lib/python3.5/site-packages/ros2pkg/api/create.py", line 71, in _create_template_file
    raise FileNotFoundError('template not found:', template_path)
FileNotFoundError: [Errno template not found:] /home/dhood/ros2_ws/install_isolated/ros2pkg/lib/python3.5/site-packages/ros2pkg/resource/package_environment/package.xml.em
```

Do I need to run it from a particular directory or anything? Or do we just need to install the resource files (I don't see how they are being installed currently).

I am posing this as a question because it might be user error, given that I am the first one to run into it when it seems that others were testing it out in https://github.com/ros2/ros2cli/pull/42

mikaelarguedas, 2018-04-02:
I used it recently without problem. I can use it in different locations

```
$ cd work/ros2/current_ws/
$ source install_debug_isolated/local_setup.bash 
$ ros2 pkg create -h
usage: ros2 pkg create [-h] [--package_format {2}] [--description DESCRIPTION]
                       [--license LICENSE]
                       [--destination-directory DESTINATION_DIRECTORY]
                       [--build-type {cmake,ament_cmake}]
                       [--dependencies DEPENDENCIES [DEPENDENCIES ...]]
                       [--maintainer-email MAINTAINER_EMAIL]
                       [--maintainer-name MAINTAINER_NAME]
                       [--cpp-node-name CPP_NODE_NAME]
                       [--cpp-library-name CPP_LIBRARY_NAME]
                       package_name

Create a new ROS2 package

positional arguments:
  package_name          The package name

optional arguments:
  -h, --help            show this help message and exit
  --package_format {2}  The package.xml format. Only version2 is supported at
                        the moment.
  --description DESCRIPTION
                        The description given in the package.xml
  --license LICENSE     The license attached to this package
  --destination-directory DESTINATION_DIRECTORY
                        Directory where to create the package directory
  --build-type {cmake,ament_cmake}
                        The build type to process the package with
  --dependencies DEPENDENCIES [DEPENDENCIES ...]
                        list of dependencies
  --maintainer-email MAINTAINER_EMAIL
                        email address of the maintainer of this package
  --maintainer-name MAINTAINER_NAME
                        name of the maintainer of this package
  --cpp-node-name CPP_NODE_NAME
                        name of the empty cpp executable
  --cpp-library-name CPP_LIBRARY_NAME
                        name of the empty cpp library
$ ros2 pkg create dummy_package --license asdf
going to create a new package
package name: dummy_package
destination directory: /home/mikael/work/ros2/current_ws
package format: 2
version: 0.0.0
description: TODO: Package description
maintainer: ['Mikael Arguedas <mikael@osrfoundation.org>']
licenses: ['asdf']
build type: ament_cmake
dependencies: []
creating folder ./dummy_package
creating ./dummy_package/package.xml
creating source and include folder
creating folder ./dummy_package/src
creating folder ./dummy_package/include/dummy_package
creating ./dummy_package/CMakeLists.txt
$ cd /tmp/
$ ros2 pkg create dummy_package --license asdf
going to create a new package
package name: dummy_package
destination directory: /tmp
package format: 2
version: 0.0.0
description: TODO: Package description
maintainer: ['Mikael Arguedas <mikael@osrfoundation.org>']
licenses: ['asdf']
build type: ament_cmake
dependencies: []
creating folder ./dummy_package
creating ./dummy_package/package.xml
creating source and include folder
creating folder ./dummy_package/src
creating folder ./dummy_package/include/dummy_package
creating ./dummy_package/CMakeLists.txt
```

dhood, 2018-04-02:
do you have a symlink install maybe?

mikaelarguedas, 2018-04-02:
Yeah likely

dhood, 2018-04-02:
could you try with non-symlink install? If it doesn't work for you either then that answers my question about why it was only broken for me (I didn't have symlink install, with symlink install it works).

dirk-thomas, 2018-04-02:
Since the `setup.py` file doesn't specify any installation for these resources and the directory doesn't have a `__init__.py` file I don't think expect the files to be installed.

mikaelarguedas, 2018-04-02:
As you pointed out there's no install rule for the resources so it's likely not going to work without symlink install. Please consider providing a PR to add the install rules and reproducible steps for this failing without the rules and passing with the new rules

Karsten1987, 2018-04-02:
Looks also like a bug to me. Sorry for that!
I can pick it up as soon as I can get to it.

dhood, 2018-04-03:
I am happy to fix it, it shouldn't be a big change, I just wanted to confirm that the fix was needed. I appreciate from the outside it seems like a strange question, but there are a number of setup.py tricks that I  am still learning e.g. there are non-obvious installation commands like `data_files` that I may have overlooked, and in the past we've also added test commands that didn't work unless you had particular python packages installed, so it was a genuine question that I had about if it was something I was missing.

I think that `data_files` is the appropriate approach for this so I will create a PR in that direction unless something else makes sense here.

closed 2018-04-10 by dhood

dhood, 2018-04-10:
updating what resulted in https://github.com/ros2/ros2cli/pull/87 (including a new setup.py trick):

`data_files` will install to `~/ros2_ws/install_isolated/ros2pkg` instead of, e.g., `~/ros2_ws/install_isolated/ros2pkg/lib/python3.5/site-packages/ros2pkg/`, where the files were originally being looked for.

They can be installed to the `site-packages` path using the `package_data` setup.py option.

When they are installed that way, they can be accessed via `pkg_resources.resource_filename('ros2pkg', 'resource/' + template_file_name)` or similar. Something I learnt is that the input to that function is not a path, per say, and [this page](http://setuptools.readthedocs.io/en/latest/pkg_resources.html#basic-resource-access) says: "Do not use os.path routines to manipulate resource paths, as they are not filesystem paths." What will be _returned_ by that function is a filesystem path, though it might be a path to a cache directory if the resources were installed in a compressed manner and so needed to be extracted first.