ros2/ros2cli #89: [ros2 pkg] add option to get package "share directory" directly

wjwwood, 2018-04-13:
Currently you can only get a package's installation prefix with this command using `ros2 pkg prefix <package name>`, but it would be convenient to be able to get at least the "share" directory of the package directly, if not others as well, i.e. `<install prefix>/share/<package name>` rather than just `<install prefix>`.

The command could either be `ros2 pkg share <package name>` or something like `ros2 pkg prefix --share <package name>` instead.

mikaelarguedas, 2018-04-13:
This could be done by using the [`get_package_share_directory`](https://github.com/ament/ament_index/blob/d9f7f2783dedc5687a5bdaf3ce5252dce5c073c5/ament_index_python/ament_index_python/packages.py#L56) function from ament_index_python

Nickolaim, 2018-06-03:
What about `ros2 pkg prefix --verbose <package name>` that is going to print package share directory as well as possibly other information in the future?

Also, can you please point me to the packages I can use for testing?

closed 2018-06-06 by dirk-thomas

yigitboracagiran, 2024-11-25:
And also you can use ament_index_cpp::get_package_share_directory if you are writing your code with C++.
[Link](https://docs.ros.org/en/rolling/p/ament_index_cpp/generated/function_get__package__share__directory_8hpp_1a3680c19cb3223de4a536289d4656ed8e.html)

PR #96 (2018-06-06): Implementation for #89
The usage is below.  Closes #89.

``` bash
$ ros2 pkg prefix ament_flake8
/home/nick/ros2_ws/install

$ ros2 pkg prefix --share ament_flake8
/home/nick/ros2_ws/install/share/ament_flake8

$ ros2 pkg prefix -h
usage: ros2 pkg prefix [-h] [--share] package_name

Output the prefix path of a package

positional arguments:
  package_name  The package name

optional arguments:
  -h, --help    show this help message and exit
  --share       show share directory for the package

```