ros2/ros2cli #91: [ros2 security] Docs: Valid keystore directory?

fkromer, 2018-05-24:
I cannot create a key for a node with `ros2 security create_key` due to an invalid keystore directory. What are valid keystore directories?

mikaelarguedas, 2018-05-24:
A valid keystore directory is a directory that contains a set of security files allowing to generate, track and sign security artifacts (it also currently contains some DDS specific files like the DDS-Security governance file).
The condition that is evaluated to determine if the keystore is valid can be found at:
https://github.com/ros2/sros2/blob/3feee9159c5304662fbe581d3673b7d3fbef77dd/sros2/api/__init__.py#L255-L264

You can create a valid keystore by running: `ros2 security create_keystore <KEYSTORE_NAME>`

HTH

fkromer, 2018-05-24:
Your answer couldn't have been better! Thx.

closed 2018-05-24 by fkromer