ros2/ros2cli #92: New Flake 8 errors

sloretz, 2018-05-31:
https://ci.ros2.org/job/ci_linux-aarch64/1426/testReport/

```
./ros2cli/plugin_system.py:63:5: F841 local variable 'e' is assigned to but never used
    except PluginException as e:
    ^

./ros2cli/plugin_system.py:68:5: F841 local variable 'e' is assigned to but never used
    except Exception as e:
    ^

./ros2cli/entry_points.py:86:9: F841 local variable 'e' is assigned to but never used
        except Exception as e:
        ^

3     F841 local variable 'e' is assigned to but never used

22 files checked
3 errors
```

dirk-thomas, 2018-05-31:
See ros2/build_cop/issues/120.

mikaelarguedas, 2018-05-31:
Duplicate of https://github.com/ros2/build_cop/issues/120#issuecomment-393310775 ?

sloretz, 2018-05-31:
Ah, I searched the build_cop repo before opening this issue, but did not find that comment. I'll leave this open since an issue is easier to find.

dirk-thomas, 2018-05-31:
The discussion on the upstream ticket doesn't look promising so we probably should just add ignores on these lines.

PR #93 (2018-06-01): ignore F841 from latest Pyflakes release
Fixes #92.
closed 2018-06-01 by dirk-thomas