ros2/ros2cli #100: Argcomplete doesn't get setup if workspace was built with colcon

dhood, 2018-06-15:
I downloaded a packaging archive built with colcon and after sourcing the `setup.bash` I don't get argcomplete functionality for `ros2cli` (whereas if the workspace is built with ament_tools I do). (@wjwwood encountered this earlier today too from source)

Does colcon not process the AMENT_ENVIRONMENT_HOOKS maybe? (maybe we're aware of this already).

I put a debug print in https://github.com/ros2/ros2cli/blob/dace3ce49154203ca270a412229a2eec8b85619f/ros2cli/completion/ros2-argcomplete.bash and confirmed that the message gets printed for old archives built with ament_tools but not for recent archives built with colcon.

dirk-thomas, 2018-06-15:
The problem is that for a plain Python package that there is nothing like `AMENT_ENVIRONMENT_HOOKS`. I see two options here:
* add a `colcon.pkg` file to the package with the following content:
  ```
  {
      "hooks": [
          "share/ros2cli/environment/ros2-argcomplete.bash"
      ]
  }
  ```
* modify `colcon-ros` adding a heuristic to look in a certain directory for scripts to source (probably `share/<pkgname>/environment`?)

In either case: not using `colcon` but building a package with the native tool (`python setup.py install`) doesn't work anyway.

Let me know which of the two options sounds better / clearer / more discoverable / more robust to you.

dirk-thomas, 2018-06-15:
The second option would probably only apply to Python and plain CMake packages. Otherwise it wouldn't be possible to distinguish the "custom" environment scripts from the "standard" environment scripts generated by `ament_cmake` (which are already being used by the package specific script.

dhood, 2018-06-15:
first option seems fine for me, most straight forward by the sounds

PR #101 (2018-06-16): add colcon.pkg file to source completion scripts
Fixes #100.
closed 2018-06-16 by dirk-thomas