ros2/ros2cli #105: ros2 node list doesn't include node namespace

dhood, 2018-06-21:
I first noticed this running a node with a remapped namespace, but it also happens if I hard-code a namespace. It happens:
- hard-coded or using static remapping
- if the node uses rclcpp or rclpy
- for a daemon using connext, fastrtps, or opensplice.

E.g.:
```
$ ros2 run demo_nodes_cpp talker  # (modified to use hard-coded namespace)
[INFO] [my_ns.talker]: Publishing: 'Hello World: 1'

$ ros2 node list
talker
```

I would consider this a bug because I expected the output to include the node's namespace, unless we are aware that this is not supported?

sloretz, 2018-06-21:
just noting ROS 1 includes the namespace

```
$ rosnode list
/gazebo
/gazebo_gui
/iris/joy_node
/iris/robot_state_publisher
/iris/roll_pitch_yawrate_thrust_controller_node
/iris/rotors_joy_interface
/rosout
```

dhood, 2018-06-21:
I think this has just never been supported. In Ardent it also didn't expand. I will re-classify this as an enhancement.

dirk-thomas, 2018-06-27:
The underlying API doesn't provide the namespace so the command line tools can't show them atm. This needs to be addressed in ros2/rmw#142 first.

PR #146 (2018-09-06): node name with namespace
Follow up of ros2/rmw#142.

Proposed to replace #143 and #144.

Fixes #105, fixes #128 and fixes ros2/ros2#569.
closed 2018-09-06 by dirk-thomas