ros2/ros2cli #109: `ros2 pkg create` traceback on windows

sloretz, 2018-06-22:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - binaries
- Version or commit hash:
  - https://ci.ros2.org/view/packaging/job/packaging_windows/1043/
- DDS implementation:
  - fastrtps, opensplice, connext
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
ros2 pkg create a_new_package
```

#### Expected behavior

Not sure, but the help text from `ros2 pkg create -h` seems to indicate this is a valid call.

#### Actual behavior

```
>ros2 pkg create a_new_package
Traceback (most recent call last):
  File "C:\dev\ros2-windows\Scripts\ros2-script.py", line 11, in <module>
    load_entry_point('ros2cli==0.4.0', 'console_scripts', 'ros2')()
  File "C:\dev\ros2-windows\Lib\site-packages\ros2cli\cli.py", line 69, in main
    rc = extension.main(parser=parser, args=args)
  File "C:\dev\ros2-windows\Lib\site-packages\ros2pkg\command\pkg.py", line 39, in main
    return extension.main(args=args)
  File "C:\dev\ros2-windows\Lib\site-packages\ros2pkg\verb\create.py", line 88, in main
    stdout=subprocess.PIPE)
  File "c:\python36\lib\subprocess.py", line 709, in __init__
    restore_signals, start_new_session)
  File "c:\python36\lib\subprocess.py", line 997, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] The system cannot find the file specified
```
#### Additional information

dhood, 2018-06-22:
Is that posted on the correct issue @dirk-thomas ?

dirk-thomas, 2018-06-22:
Yes: https://github.com/ros2/ros2cli/blob/1d24ad58e32846522d71c52380d10ce09c2983d9/ros2pkg/ros2pkg/verb/create.py#L86-L88

dhood, 2018-06-22:
woah, I wasn't expecting that 😄

wjwwood, 2018-06-22:
Based on this, `ros2create` maybe should depend on `git` and `git` should be added to the install instructions for Windows binaries:

https://github.com/ros2/ros2/wiki/Windows-Install-Binary#installing-prerequisites

wjwwood, 2018-06-22:
Or we should refactor this to not need `git`, or use it if available, but not require it.

wjwwood, 2018-06-22:
Also, used just a few lines below:

https://github.com/ros2/ros2cli/blob/1d24ad58e32846522d71c52380d10ce09c2983d9/ros2pkg/ros2pkg/verb/create.py#L99-L101

dirk-thomas, 2018-06-22:
I would rather not add a dependency on `git` but fall back to a default value if it is not available.

mikaelarguedas, 2018-06-22:
> I would rather not add a dependency on git but fall back to a default value if it is not available.

:+1: the catkin equivalent uses `getpass.getuser` by default:

https://github.com/ros-infrastructure/catkin_pkg/blob/1c6ca153b43d6c8edec23f0f0059258e26a28086/src/catkin_pkg/package_templates.py#L77

sloretz, 2018-06-25:
> I assume you have git installed? 

Git isn't installed

PR #111 (2018-06-25): ros2pkg: avoid requiring git
Fixes #109.

I didn't run CI for this but ran the package tests as well as the actual verb locally.
closed 2018-06-25 by dirk-thomas