ros2/ros2cli #110: `ros2 topic echo` Crashes when using opensplice

sloretz, 2018-06-22:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - binaries
- Version or commit hash:
  - https://ci.ros2.org/view/packaging/job/packaging_windows/1043/
- DDS implementation:
  - rmw_opensplice_cpp
- Client library (if applicable):
  -  N/A

#### Steps to reproduce issue

```cmd
REM Make sure daemon is using opensplice
set RMW_IMPLEMENTATION=rmw_opensplice_cpp
ros2 daemon stop
ros2 daemon start
ros2 run demo_nodes_cpp talker

REM in another terminal
ros2 topic echo chatter
```

#### Expected behavior

Messages would be echo'd on second terminal

#### Actual behavior

```
>ros2 topic echo chatter
========================================================================================
Context     : DDS::WaitSet::detach_condition
Date        : Fri Jun 22 16:01:21 Pacific Daylight Time 2018
Node        : DESKTOP-S5KSNBT
Process     : python.exe <4384>
Thread      : 1544
Internals   : WaitSet.cpp/456/6.7.180404OSS/5926d257/5926d257/-1
----------------------------------------------------------------------------------------
Report      : Precondition not met: This ReadCondition is being deleted.
Internals   : DDS::OpenSplice::ReadCondition::wlReq_detachFromWaitset/ReadCondition.cpp/176/4/1529708482.015405000
Failed to wait on wait set: failed to detach guard condition, at C:\J\workspace\packaging_windows\ws\src\ros2\rmw_opensplice\rmw_opensplice_cpp\src\rmw_wait.cpp:282, at C:\J\workspace\packaging_windows\ws\src\ros2\rcl\rcl\src\rcl\wait.c:647
```

#### Additional information

sloretz, 2018-06-22:
Probably duplicate of ros2/ros2cli#107

dirk-thomas, 2018-06-22:
It isn't a duplicate since the error happens at a different condition. Please try to update ros2/rmw_opensplice#235 to add the same handling for this case (and probably also for all other condition).

mikaelarguedas, 2018-06-22:
Similar to https://github.com/ros2/ros2cli/issues/107, the workaround from https://github.com/ros2/rmw_opensplice/pull/235 may apply to this bug as well

dirk-thomas, 2018-06-27:
Closing since it has been addressed by ros2/rmw_opensplice#235. The remaining warning is ticketed in ros2/rmw_opensplice#237.

closed 2018-06-27 by dirk-thomas