ros2/ros2cli #112: `ros2 topic echo` deserialization failed using connext

sloretz, 2018-06-25:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - binaries
- Version or commit hash:
  - https://ci.ros2.org/view/packaging/job/packaging_windows/1046/
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Start a talker executable in one terminal
```bat
set RMW_IMPLEMENTATION=rmw_connext_cpp
ros2 run demo_nodes_cpp talker
```

Echo the topic in another terminal
```bat
set RMW_IMPLEMENTATION=rmw_connext_cpp
REM make sure the daemon uses connext
ros2 daemon stop
ros2 daemon start

ros2 topic echo /chatter
```

#### Expected behavior

Every message published would be echoed on  the console

#### Actual behavior

The first message is printed, but then an error message `deserialize from cdr buffer failed` appears. The command never prints another message despite the talker continuing to publish.

```bat
C:\dev\ros2-windows>ros2 topic echo /chatter
RTI Data Distribution Service Evaluation License issued to OSRF (OSRF01) dthomas@osrfoundation.org For non-production use only.
Expires on 5-Nov-2018 See www.rti.com for more information.
data: 'Hello World: 50'

deserialize from cdr buffer failed
```

#### Additional information
Additionally, the command `ros2 topic echo /chatter` cannot be interrupted using `CTRL` + `c`. I had to kill the process in the task manager.

dhood, 2018-06-25:
I've seen this error also get printed in our nightlies e.g.
```
2:  - /home/rosbuild/ci_scripts/ws/build/test_communication/test_publisher_subscriber_cpp BoundedArrayNested
2: cdr stream doesn't contain data
2: deserialize from cdr buffer failed
2: [ERROR] [rclcpp]: could not deserialize serialized message on topic '/test_message_BoundedArrayNested': can't convert cdr stream to ros message, at /home/rosbuild/ci_scripts/ws/src/ros2/rmw_connext/rmw_connext_cpp/src/rmw_take.cpp:185, at /home/rosbuild/ci_scripts/ws/src/ros2/rcl/rcl/src/rcl/subscription.c:258
```
(appears 17 times in various tests in https://ci.ros2.org/view/nightly/job/nightly_linux_release/854/consoleFull, for example)

dirk-thomas, 2018-06-25:
Can you clarify why the rmw impl. of the daemon is relevant here? The `echo` isn't being processed by the daemon but using `rclpy` in the `echo` verb itself.

sloretz, 2018-06-25:
>Can you clarify why the rmw impl. of the daemon is relevant here? 

I don't know if making sure the daemon is on the right RMW implementation is important or not. I included it because it is part of the sequence of commands I ran while testing `ros2 topic`.

mikaelarguedas, 2018-06-26:
I can reproduce this on xenial,

```
$ RMW_IMPLEMENTATION=rmw_connext_cpp ros2 topic echo /chatter
RTI Data Distribution Service Evaluation License issued to OSRF (OSRF01) dthomas@osrfoundation.org For non-production use only.
Expires on 5-Nov-2018 See www.rti.com for more information.
data: 'Hello World: 1'

deserialize from cdr buffer failed

```
 First message is received correctly but then the error message is printed (before receiving a second message).

Process exits cleanly with Ctrl+C for me.

Not using the Direct node construct in ros2cli fixes the issue for me (https://github.com/ros2/ros2cli/pull/116)
Suppressing the spinning from the DirectNode fixes the issue as well, but this cannot be done as most other tools need it to behave that way.

The snippet linked by @dhood above can be reproduced on the connext receiver side (regardless of the RMW implementation used on the publisher side).
Once the publisher is shutdown, the subscriber receives the notification that the matched writer is shutting down and for some obscur reason enters take (and fails). That issue is unrelated to ros2cli so I'll ticket it on another repo.

closed 2018-06-26 by mikaelarguedas

mikaelarguedas, 2018-06-26:
> That issue is unrelated to ros2cli so I'll ticket it on another repo.

There's already a PR in to address the remaining issue: https://github.com/ros2/rmw_connext/pull/297

PR #116 (2018-06-26): pass actual node object to subscriber function
Opening for visibility until a better workaround is found

Right now the spinning in DIrectNode causes #112 on Connext.

This PR hackily moves the logic to create and destroy the node in the echo verb itself as there is no need for a bootstrap phase for the node as it doesn't need to aggregate network information.

Fixes #112