ros2/ros2cli #114: Flaky crash at shutdown using subscriber_old_school

sloretz, 2018-06-26:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - binaries
- Version or commit hash:
  - https://ci.ros2.org/view/packaging/job/packaging_windows/1046/
- DDS implementation:
  - opensplice
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

Run subscriber old school
```
ros2 run examples_rclpy_minimal_subscriber subscriber_old_school
```

`CTRL` + `c`. Sometimes there is a crash instead of a clean shutdown.

#### Expected behavior

Executable would shut down cleanly.

#### Actual behavior
```
C:\dev\ros2-windows>ros2 run examples_rclpy_minimal_subscriber subscriber_old_school
[INFO] [minimal_subscriber]: I heard: "Hello World: 64"
[INFO] [minimal_subscriber]: I heard: "Hello World: 65"
[INFO] [minimal_subscriber]: I heard: "Hello World: 66"
[INFO] [minimal_subscriber]: I heard: "Hello World: 0"
[INFO] [minimal_subscriber]: I heard: "Hello World: 1"
[INFO] [minimal_subscriber]: I heard: "Hello World: 2"
[INFO] [minimal_subscriber]: I heard: "Hello World: 0"
[INFO] [minimal_subscriber]: I heard: "Hello World: 1"
[INFO] [minimal_subscriber]: I heard: "Hello World: 2"
========================================================================================
Context     : DDS::WaitSet::wait
Date        : Mon Jun 25 17:50:55 Pacific Daylight Time 2018
Node        : DESKTOP-S5KSNBT
Process     : python.exe <3116>
Thread      : 2992
Internals   : WaitSet.cpp/252/6.7.180404OSS/5926d257/5926d257/-1
----------------------------------------------------------------------------------------
Report      : Already deleted: Object is already deleted.
Internals   : DDS::OpenSplice::CppSuperClass::check/CppSuperClass.cpp/232/9/1529974255.092807600
----------------------------------------------------------------------------------------
Report      : Precondition not met: Waitset is already deleted
Internals   : u_waitsetWaitAction2/u_waitset.c/333/782/1529974255.102913900
```

#### Additional information

nuclearsandwich, 2018-06-27:
Building from source with opensplice on Windows and I was not able to reproduce this behavior running and stopping this process multiple times with various run intervals between.

claireyywang, 2019-09-25:
closing as the issue does not seem to be reproducible, will reopen if it persists in latest versions.

closed 2019-09-25 by claireyywang