ros2/ros2cli #127: ros2 topic fail to echo PointCloud2 msg

yechun1, 2018-07-25:
Issue discussing here: https://discourse.ros.org/t/image-and-pointcloud2-performance-on-ros2/5391

When I try to use big data message by python code such as Camera Image or PointCloud2 on ROS2, I found the performance is terrible.
Problem:
    Launch ros2_intel_realsense 2 by realsense_ros2_camera, and subscribe “/camera/depth/color/points” by “ros2 topic echo /camera/depth/color/points”. It will cost dozens of seconds to print out the PointCloud2 (640x480) data. On ros 1, rostopic shows data immediately.

yechun1, 2018-07-25:
PR https://github.com/ros2/ros2cli/pull/126 to fix this issue, please help review.

mikaelarguedas, 2018-08-01:
Fixed by #126 , thanks @yechun1 for the fix

closed 2018-08-01 by mikaelarguedas