ros2/ros2cli #128: 'ros2 param list' stuck when node create with namespace

realdream, 2018-07-31:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - from source
- Version or commit hash:
  - release bouncy
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp
#### Steps to reproduce issue

```
ros2 run demo_nodes_cpp talker __ns:=/demo
ros2 param list
```
#### Actual behavior
the 'ros2 param list' stuck forever with out output any message

sloretz, 2018-08-01:
Thanks for the bug report @realdream . I can reproduce `ros2 param list` being stuck forever and being unable to get/set parameters using bouncy patch release 1 installed from debs.

Output when trying to get or set a parameter on a node with a namespace set.

```
developer@bbd2c434522d:~$ ros2 param set talker foo true
Wait for service timed out
developer@bbd2c434522d:~$ ros2 param get talker foo
Wait for service timed out
developer@bbd2c434522d:~$ ros2 param get /demo/talker foo
Node not found
developer@bbd2c434522d:~$ ros2 param set /demo/talker foo true
Node not found
```

Services and topics

```
developer@bbd2c434522d:~$ ros2 service list
/demo/talker/describe_parameters
/demo/talker/get_parameter_types
/demo/talker/get_parameters
/demo/talker/list_parameters
/demo/talker/set_parameters
/demo/talker/set_parameters_atomically
developer@bbd2c434522d:~$ ros2 topic list
/clock
/demo/chatter
/demo/parameter_events
```

Possibly related to ros2/ros2cli#105 ? `list` gets all node names from `ros2node`'s api
https://github.com/ros2/ros2cli/blob/be38ea405388672c42eeb02fc9ccbffb6ed6945d/ros2param/ros2param/verb/list.py#L42-L43

Then it creates a client with the node name assuming it will be a single token.
https://github.com/ros2/ros2cli/blob/be38ea405388672c42eeb02fc9ccbffb6ed6945d/ros2param/ros2param/verb/list.py#L55-L57

mjcarroll, 2018-08-17:
I agree with your assessment @sloretz.  This is going to be dependent on the `rmw` returning a list of namespaced names, in addition to just the node names.

PR #146 (2018-09-06): node name with namespace
Follow up of ros2/rmw#142.

Proposed to replace #143 and #144.

Fixes #105, fixes #128 and fixes ros2/ros2#569.
closed 2018-09-06 by dirk-thomas