ros2/ros2cli #132: [ros2topic] enable ros2topic hz delay command

yechun1, 2018-08-13:
Reference to discussing ["rostopic command for ROS2"](https://discourse.ros.org/t/rostopic-command-for-ros2/2565/4)

ROS topic commands are very useful for simple performance tests such as hz/delay/bw, but those tools are not supported in ros2 topic right now, if there is no body work on there. I would like to contribute to port ROS topic(hz/delay/bw) to ROS2.

Todo list:
- [ ros2 topic hz] 
- [ ros2 topic delay]
- [ ros2 topic bw]

dirk-thomas, 2018-08-13:
Great! Let use know if you have any questions.

yechun1, 2018-10-08:
While porting bw command, did not find the api to get size of the subscribed msg. Could anyone help? 

ROS use `AnyMsg` and `len(data._buff)` to get the size (line to https://github.com/ros/ros_comm/blob/melodic-devel/tools/rostopic/src/rostopic/__init__.py#L440). But how to replace on ROS2?

josephduchesne, 2018-10-08:
I'm not sure if there is a canonical size of a ros message any more, especially exposed in python. 

Rclpy now hands serialization off to the cpp rcl then rmw layer from what I can tell.

https://github.com/ros2/demos/pull/185 and the other PRs linked therin add rmw+rcl+rclcpp support for returning the raw serialized message in a raw type callback, but I don't think this is exposed in rclpy (yet?).

I wonder which would be easier or better:
- Port bw to C++ and using the existing raw function (Can ros2cli verbs be written in C++?)
- Write a purpose built C++ glue function to get just this data and invoke it from the python side
- Add raw=False keyword arg to rclpy Node.create_subscription and all the way down to support returning a byte list subscription in python.

I may take a crack at the last option there and if it turns out to be reasonable, I'll submit a PR and link it here, otherwise I'll try to explain what trouble I ran into here.

josephduchesne, 2018-10-08:
I've created PR ros2/rclpy#242 that needs review for design, and some touching up to work towards providing a rospy implementation of raw message subscriptions like roscpp has thanks to ros2/demos#185

LucidOne, 2018-10-24:
One complementary approach I have been thinking about is updating the metrics/counters in [`Publisher::publish()`](https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/publisher.hpp#L189) since the data is there already. It could be accessed via `Publisher::get_bw()` and used with a diagnostic aggregator implementation.

josephduchesne, 2019-01-10:
ros2/rclpy#242 is now merged, so `ros2 topic bw` should be fairly straight forward to implement now.

Something like
```
subscription = self.create_subscription(..., raw=True)
```
then in the callback, `len(msg)` will be the serialized byte length required.

dirk-thomas, 2019-01-10:
Actually subscribing to the messages to then calculate their length sounds highly inefficiency to me.

yechun1, 2019-01-11:
`len(msg)` to get the length of subscribing message is very similar with what ROS1 ` len(msg._buff)` does. It might be inefficiency, but at least, could make ros2topic bw work. Could we first enable ros2topic bw feature and later improve if we found the better APIs?

yechun1, 2019-01-11:
thanks @josephduchesne .
Here is patch ready for review https://github.com/ros2/ros2cli/pull/190

LucidOne, 2019-01-11:
@yechun1 What are your thoughts on putting the bandwidth metrics in the publisher?

mjcarroll, 2019-01-11:
@LucidOne I would be inclined to believe that would be the most efficient approach, but would certainly imply changes deeper in the stack (at least `rclpy` or `rclcpp`, but potentially deeper).  @dirk-thomas would you be in favor of working with the existing implementation in `ros2cli` with the understanding that it would be further enhanced down the road?

dirk-thomas, 2019-01-11:
The current approach of the PR is ok to be merged. If anyone would continue the effort to make it more efficient afterwards that would be good though.

LucidOne, 2019-01-11:
I'm more or less :+1: for merging #190 and improving things later.

In addition to debugging, I think there is also a programmatic need for bandwidth utilization information, especially in multi-robot systems. Does it make sense to add a publisher here, or in a separate node/API? If it is a separate node, should the bandwidth calculation be pushed into a module? An API may be be more efficient for the diagnostic aggregator to monitor topics exceeding bandwidth limits.

yechun1, 2019-04-18:
`todo list` all done, close this issue 
 [ ros2 topic hz] (#133)
  [ ros2 topic delay] (#139)
  [ ros2 topic bw] (#190)

closed 2019-04-18 by yechun1