ros2/ros2cli #148: Cannot start the daemon if no networks are up

clalancette, 2018-09-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 (ARM64, on a Rock64 board)
- Installation type:
  - Source
- Version or commit hash:
  - master
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

1.  Compile ROS2 from source: `colcon build --event-handlers console_cohesion+`
1.  Source the workspace: `source install/setup.bash`
1.  Take down any active network interfaces, so that only loopback is running.
1.  Run `ros2 topic list` (or really, any other command that causes the ros2 daemon to be started)

#### Expected behavior
`ros2 topic list` returns with an empty list (or possibly just `/parameter_events`).

#### Actual behavior
`ros2 topic list` hangs "forever", or at least as long as I was willing to wait (10 minutes or so).

#### Additional information
Hitting `Ctrl-C` when `ros2 topic list` hangs like this results in a Segmentation Fault (less problematic, but still not nice).

I suspect a bug in the daemon because the following sequence works:

1.  Bring the network back up.
1.  Run `ros2 topic list` (this returns the expected empty list, and launches the daemon)
1.  Bring the network back down.
1.  Now `ros2 topic list` seems to work.

mjcarroll, 2018-09-11:
It seems like it should actually warn the user in the case that no networks are detected rather than just returning an empty list.

mikaelarguedas, 2018-09-11:
This has been an issue for fastrtps for as long as I can remember (needing a network to be up for Discovery and data exchange to work even on loopback). I believe it is wider than the daemon. There are a few discourse posts and FastRTPS issues on that topic I believe

clalancette, 2019-01-03:
Retested this with the latest code on master, and things work as expected, so closing this out.

closed 2019-01-03 by clalancette