ros2/ros2cli #151: 'ros2 param list' stuck on node 'test'

MarcTestier, 2018-10-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - 2049b2f
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
ros2 param list
```

#### Expected behavior
Show the list of parameters.

#### Actual behavior
Doesn't output anything and stay stuck forever.

#### Additional information
I checked the file `list.py` in `ro2cli/ros2param/ros2param/verb` and added a few print. I can see the command is getting stuck at line 82 (second line here): 
```
for node_name in node_names:
    rclpy.spin_until_future_complete(node, futures[node_name])
```

It loop first on some of the nodes that I launched : `rviz`, `static_transform_publisher` and `urg_node`, and stay stuck on a node I didn't launched (or not directly launched) : `test`.
The full `node_name` variable being `NodeName(name='test', namespace='/', full_name='/test')`

mjcarroll, 2018-10-11:
Which `urg_node` are you using, this one? https://github.com/bponsler/urg_node

dirk-thomas, 2018-10-11:
What does `ros2 node list` report? If it contains "stall" nodes maybe try `ros2 daemon stop` (if will restart itself transparently on the next usage).

MarcTestier, 2018-10-12:
> Which `urg_node` are you using, this one? https://github.com/bponsler/urg_node

I'm using this version of the urg_node : https://github.com/hopetechnik/urg_node/tree/ros2-devel, which I'm currently modifying it to add reconfiguration of the parameters using parameter events.

> What does `ros2 node list` report? If it contains "stall" nodes maybe try `ros2 daemon stop` (if will restart itself transparently on the next usage).

`ros2 node list` report : 
```
/urg_node
/test
/test
```
All I launched was `urg_node`, I don't know where this 2 `test` nodes come from.
I tried `ros2 deamon stop`, it said `The daemon has been stopped` but didn't change anything to the problem.

MarcTestier, 2018-10-12:
Ok, I found where the test node where launched from. It comes from the `diagnostic_updater` package which is a dependency of `urg_node`.
When the `urg_node` create a `diagnostic_updater::Updater`, it will create 2 nodes named `test`, can see it here : https://github.com/bponsler/diagnostics/blob/ros2-devel/diagnostic_updater/include/diagnostic_updater/diagnostic_updater.h#L374.

I don't know how good is the port or if it really follows "ROS 2 good practices", which probably changed while the package was ported, but I know someone else is looking into porting the package to ROS 2 : https://discourse.ros.org/t/diagnostic-aggregator-and-diagnostic-updater-porting-to-ros2/6382.

So I don't know if it really is a bug or if it just comes from something outdated ROS 1 practices in the package.

clalancette, 2026-11-14:
Given how long this has been open, and the fact it was probably a conflict, I'm going to close this one out.  Thanks for reporting.

closed 2026-11-14 by clalancette