ros2/ros2cli #152: ros2 param set, on_parameter_event called twice

MarcTestier, 2018-10-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - 2049b2f
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
ros2 param set /parameter_events plop 42
```

#### Expected behavior
Call `on_parameter_event` once with the new/changed/deleted parameters
#### Actual behavior
Call `on_parameter_event` twice, the first time with no parameters then with the new/changed/deleted parameters
#### Additional information
When I set a parameter using the parameter client in a node, `on_parameter_event` is called once, as it should be, but when I set a parameter using command line, the `on_parameter_event` is called twice. And the first time it's triggered, there is no parameters inside.
The output using a node : 
```
[INFO] [parameter_events]: 
Parameter event:
 new parameters:
  foo
 changed parameters:
 deleted parameters:
```
The output using command line `ros2 param set /parameter_events plop 42` : 
```
[INFO] [parameter_events]: 
Parameter event:
 new parameters:
 changed parameters:
 deleted parameters:

[INFO] [parameter_events]: 
Parameter event:
 new parameters:
  plop
 changed parameters:
 deleted parameters:
```
I created a simple/minimal file to test this : [here](https://github.com/MarcTestier/test_repo/blob/master/cpp_test_repo/src/parameter_events_async.cpp), based on the `demos` pkg.

I'm not 100% sure whether the issue should be here or in rclcpp, so I apologize if it should be in the other repo.

wjwwood, 2018-10-18:
That is strange, but I didn't have a chance to reproduce it yet. I don't have an ideas off the top of my head as to why it would be called twice. I don't think it's expected behavior.

Have you tried creating a C++ program that just sets the parameter on the node from another process (rather than from within the same node/process)? If it also doesn't get called twice in that case I'd say it's definitely an issue with this tool, otherwise it might be related to where the node is being set from.

MarcTestier, 2018-10-18:
> Have you tried creating a C++ program that just sets the parameter on the node from another process (rather than from within the same node/process)? If it also doesn't get called twice in that case [...]

If you mean having a NODE 1 set param on NODE 2, then yes I did and `on_parameter_event` is called once.

clalancette, 2026-11-14:
I couldn't reproduce this on currently rolling, so closing this out.  Please feel free to open a new issue if you are still having the problem.

closed 2026-11-14 by clalancette