ros2/ros2cli #157: Add Better Documentation

davetcoleman, 2018-11-08:
This is a rather core library of ROS2 so should set a good example of documentation to the ROS community. It will only take a few minutes to:

1) Add a description at the top of the Github main page where it says "No description, website, or topics provided."

2) Add a README that includes a CI badge and a link to relevant documentation such as the [Discourse discussion](https://discourse.ros.org/t/ros-graph-information-tools-implementation-discussion/674/39) or rosindex

artivis, 2019-07-08:
In the meantime I've put together a small [CLI Cheats Sheet](https://github.com/artivis/ros2_cheats_sheet/blob/master/cli/cli_cheats_sheet.pdf).

AnandShastry, 2019-11-10:
Need proper documentation for all CLI. 
for example:
average rate: ??
	min: ?? max: ?? std dev: ?? window: ??

All above should be explained with example.

clalancette, 2022-10-28:
I'll say that the original request here was fulfilled long ago, so I'll close this out.

We do need better documentation overall, but we can track that separately.

closed 2022-10-28 by clalancette