ros2/ros2cli #164: `ros2 node list --spin-time` inconsistent behavior if daemon is started or stopped

sloretz, 2018-12-10:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Bionic
- Installation type:
  - debs
- Version or commit hash:
  - crystal pre-release
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Time how long it takes to list nodes with `--spin-time` and the daemon stopped

```
ros2 daemon stop
time ros2 node list --spin-time 15
```

Repeat with the daemon started

```
ros2 daemon start
time ros2 node list --spin-time 15
```

#### Expected behavior

`ros2 node list --spin-time 15` would always take about 15 seconds to complete

#### Actual behavior

With the daemon stopped it takes the expected time

```
$ ros2 daemon stop
The daemon has been stopped
$ time ros2 node list --spin-time 15

real    0m15.336s
user    0m0.284s
sys     0m0.056s
```

With the daemon started it completes very quickly

```
$ ros2 daemon start
The daemon is already running
$ time ros2 node list --spin-time 15

real    0m0.322s
user    0m0.280s
sys     0m0.040s
```

#### Additional information

There are two stragegies: direct and daemon. 
https://github.com/ros2/ros2cli/blob/af65db2d95fbbd436308d7fd363c6545342496ad/ros2cli/ros2cli/node/strategy.py#L26-L30

`DirectNode` checks `spin_time`
https://github.com/ros2/ros2cli/blob/af65db2d95fbbd436308d7fd363c6545342496ad/ros2cli/ros2cli/node/direct.py#L39-L43

`DaemonNode` does not. It would need to check spin time and spin before calling `get_node_names_and_types()`

dirk-thomas, 2018-12-11:
Why should the command line tool wait 15 seconds when the daemon is being used? Maybe the help text of the option should just be updated stating that it only affects the case when a new node needs to be created but not when the daemon is available.

sloretz, 2018-12-11:
> Why should the command line tool wait 15 seconds when the daemon is being used?

It's possible the dameon has been running for less time than `--spin-time`.  I'm fine with a documentation fix though.

clalancette, 2026-11-14:
This was documented ages ago, so closing this out.

closed 2026-11-14 by clalancette