ros2/ros2cli #165: ros2 topic hz affects subscription rates

hidmic, 2018-12-11:
## Bug report

**Required Info:**
- Operating System:
  - Ubuntu Bionic 18.04 (as in [Nvidia OpenGL Docker image](https://hub.docker.com/r/nvidia/opengl/))
- Installation type:
  - Binary (debs)
- Version or commit hash:
  - Crystal pre-release.
- DDS implementation:
  - ADLink OpenSplice 6.9
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Bring up the nodes as described on the [QoS tutorial](https://index.ros.org/doc/ros2/Quality-of-Service/). For reference, that means:
```sh
RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 run image_tools cam2image
```
and
```sh
RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 run image_tools showimage
```
in separate terminals.

Now, run:

```
RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 topic hz /image
```

#### Expected behavior
Publish rates for the `/image` topic shown correctly with no observable effect on publisher or subscriber.
#### Actual behavior
Subscription rate for the `image_tools showimage` command slows down considerably. `ros2 topic hz` command output reflects this. `image_tool cam2image` keeps publishing as before but some warnings start to show up:

```sh
========================================================================================                           
Report      : WARNING                                                                                              
Date        : 2018-12-11T12:16:04-0300                                                                             
Description : thread tev failed to make progress                                                                   
                                                                                                                   
Node        : abydos                                                                                               
Process     : cam2image <4769>                                                                                     
Thread      : lease 7f6d680dd700                                                                                   
Internals   : 6.9.181126OSS///ddsi2/q_servicelease.c/131/0/1544541364.949280247/0
========================================================================================                           
Report      : WARNING                                                                                              
Date        : 2018-12-11T12:16:04-0300                                                                             
Description : thread xmit.user failed to make progress                                                             
                                                                                                                   
Node        : abydos                                                                                               
Process     : cam2image <4769>                                                                                     
Thread      : lease 7f6d680dd700                                                                                   
Internals   : 6.9.181126OSS///ddsi2/q_servicelease.c/131/0/1544541364.949362717/0
```
#### Additional information
It does not happen when using Fast-RTPS as RMW of choice.

wjwwood, 2018-12-12:
I wouldn't expect that this should impact rate very much, but I'm not surprised it impacts it some, because `hz` has to subscribe to the topic in order to calculate the frequency, meaning all the data (in this case an image) is sent just to see how frequently it is sent. That will put more load on the system.

But those warnings point to something more abnormal happening.

Can you quantify the slowdown amount?

hidmic, 2019-01-03:
Hmm. I can no longer reproduce with Crystal released (as opposed to pre-released) debians.

claireyywang, 2019-09-23:
Close ticket since the issue does not persist in later versions.

closed 2019-09-23 by claireyywang

monticle, 2024-07-17:
Using a correct gateway address solved my problem.