ros2/ros2cli #175: [ros2lifecycle] get --transitions raises an exception

hidmic, 2018-12-12:
## Bug report

**Required Info:**
- Operating System:
  - Ubuntu Bionic 18.04 (as in [Nvidia OpenGL Docker image](https://hub.docker.com/r/nvidia/opengl/))
- Installation type:
  - Binary (debs)
- Version or commit hash:
  - Crystal pre-release.
- DDS implementation:
  - ADLink OpenSplice 6.9
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Start a managed talker node:

```sh
ros2 run lifecycle lifecycle_talker
```
Now attempt to get the current lifecycle state plus possible transitions:

```sh
ros2 lifecycle get /lc_talker --transitions
```

#### Expected behavior
Current state is shown along with the possible transitions, as documented.
#### Actual behavior
It throws an `AttributeError` exception and no transition is listed.

```sh
Traceback (most recent call last):
  File "/opt/ros/crystal/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.6.1', 'console_scripts', 'ros2')()
  File "/opt/ros/crystal/lib/python3.6/site-packages/ros2cli/cli.py", line 69, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/crystal/lib/python3.6/site-packages/ros2lifecycle/command/lifecycle.py", line 40
, in main
    return extension.main(args=args)
  File "/opt/ros/crystal/lib/python3.6/site-packages/ros2lifecycle/verb/get.py", line 93, in main
    .format_map(locals()))
AttributeError: 'TransitionDescription' object has no attribute 'label'
```

sloretz, 2018-12-12:
Duplicate of ros2/ros2cli#170 fixed by ros2/ros2cli#174

hidmic, 2018-12-12:
Argh! I checked opened ones, not closed ones. Thanks @sloretz !

closed 2018-12-12 by hidmic