ros2/ros2cli #178: [ros2param] ros2 param list requires fully qualified node name

sloretz, 2018-12-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu bionic
- Installation type:
  -  debs
- Version or commit hash:
  - crystal prerelease
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

1. Save the following as `demo_params.yaml`
   ```
   talker:
       ros__parameters:
           some_int: 42
           a_string: "Hello world"
           some_lists:
               some_integers: [1, 2, 3, 4]
               some_doubles : [3.14, 2.718]
   ```
1. Launch a node with those parameters
   ```
   ros2 run demo_nodes_cpp talker __params:=demo_params.yaml
   ```
1. List the parameters using a fully qualified node name
   ```
   ros2 param list /talker
   ```
1. List the parameters using a non fully qualified node name
   ```
   ros2 param list talker
   ```

#### Expected behavior

Both commands would list the parameters on the talker node. In the first case the name is fully qualified already so the node to list is unambiguous. In the second case the namespace of the CLI tool is `/` by default, so it should list the params for `/` + `talker`. 

#### Actual behavior

The non-fully qualified version can't find the node.

```bash 
$ ros2 param list /talker
  a_string
  some_int
  some_lists.some_doubles
  some_lists.some_integers
$ ros2 param list talker
Node not found
```

#### Additional information

Karsten1987, 2018-12-13:
we could address this in https://github.com/ros2/ros2cli/pull/179
opinions?

Karsten1987, 2018-12-13:
we could address this in #179
opinions?

closed 2018-12-13 by sloretz