ros2/ros2cli #180: [ros2service] Service type autocompletion for OpenSplice brings wrapping type

hidmic, 2018-12-13:
## Bug report

**Required Info:**
- Operating System:
  - Ubuntu Bionic 18.04 (as in [Nvidia OpenGL Docker image](https://hub.docker.com/r/nvidia/opengl/))
- Installation type:
  - Binary (debs)
- Version or commit hash:
  - Crystal pre-release.
- DDS implementation:
  - ADLink OpenSplice 6.9
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

On a terminal, start an example service server:

```sh
RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 run demo_nodes_cpp add_two_ints_server
```

On another terminal, attempt to autocomplete the type by pasting the following command and pressing TAB:

```sh
RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 service call /add_two_ints
```

#### Expected behavior
Autocompletion brings the right service type i.e. `example_interfaces/AddTwoInts`.

#### Actual behavior
Autocompletion brings the _wrapping_ type i.e. `example_interfaces/Sample_AddTwoInts`.

dirk-thomas, 2018-12-13:
You have a daemon process running which is not using OpenSplice. Try to stop the daemon (`ros2 daemon stop`) and then call `RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 service call /add_two_ints` again.

hidmic, 2018-12-13:
That's right. I would've sweared I had stopped it. Thanks :)

closed 2018-12-13 by hidmic