ros2/ros2cli #182: [ros2{topic,service}] the --include-hidden-* option should be an option of list verb not command

wjwwood, 2018-12-14:
## Bug report

**Required Info:**

- Operating System:
  - any
- Installation type:
  - any
- Version or commit hash:
  - Crystal release candidate
- DDS implementation:
  - any
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
$ ros2 topic -h
$ ros2 topic list -h
```

#### Expected behavior

Options that only affect `list` verb would show up in its own help and be passed after the verb. At the very least it should be seen as an option when you do `ros2 topic list -h` (for example).

#### Actual behavior

Option only appears in command help, e.g. `ros2 topic -h`.

#### Additional information

I'm not sure why it's this way in the first place, possibly because the option used to affect the echo/call verbs. But right now it's definitely confusing and difficult to discover this option.

rarvolt, 2019-01-06:
I have moved the `--include-hidden-topic` to the `list` verb. Are there any other issues like this that I'm not aware of, or it's just that?

dirk-thomas, 2019-02-09:
Looking at the code each and every verb under the `topic` command uses the `--include-hidden-topics` argument either for the topic name completion or `list` which determines which topic names to print:
* [delay](https://github.com/ros2/ros2cli/blob/2f9e81d86b06e1b5f397bdfdbd33493cbc5f27c1/ros2topic/ros2topic/verb/delay.py#L69)
* [echo](https://github.com/ros2/ros2cli/blob/2f9e81d86b06e1b5f397bdfdbd33493cbc5f27c1/ros2topic/ros2topic/verb/echo.py#L51)
* [hz](https://github.com/ros2/ros2cli/blob/2f9e81d86b06e1b5f397bdfdbd33493cbc5f27c1/ros2topic/ros2topic/verb/hz.py#L74)
* [info](https://github.com/ros2/ros2cli/blob/2f9e81d86b06e1b5f397bdfdbd33493cbc5f27c1/ros2topic/ros2topic/verb/info.py#L28)
* [list](https://github.com/ros2/ros2cli/blob/2f9e81d86b06e1b5f397bdfdbd33493cbc5f27c1/ros2topic/ros2topic/verb/list.py#L37)
* [pub](https://github.com/ros2/ros2cli/blob/2f9e81d86b06e1b5f397bdfdbd33493cbc5f27c1/ros2topic/ros2topic/verb/pub.py#L36)

So moving the argument to only the `list` verb breaks that functionality for all the other verbs. Therefore I don't see what should be done here? @wjwwood Please clarify to acceptance criteria of this ticket.

dirk-thomas, 2019-02-13:
@wjwwood Can you please comment here?

wjwwood, 2019-02-13:
You could either leave `--include-hidden-topics` in the topic command (so the other verbs can use it for tab completion) or you could have tab completion always list hidden topics. I don't mind which, but for me the acceptance criteria is that `--include-hidden-topics` should be an option for the `list` verb, regardless of whether or not it is duplicated in the command as well.

dirk-thomas, 2019-02-14:
> `--include-hidden-topics` should be an option for the `list` verb

Implemented in #196.

PR #196 (2019-02-14): duplicate --include-hidden-topics in list verb
Closes #182.
closed 2019-02-14 by dirk-thomas