ros2/ros2cli #183: [ros2{node,topic,service}] OpenSplice doesn't display topics/services

mjcarroll, 2018-12-14:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10, Visual Studio 2017
- Installation type:
  - Source
-  Version or commit hash:
  - Crystal ros2.repos (https://raw.githubusercontent.com/ros2/ros2/f5d204bb671ac875872693b405ba86a8c2abb251/ros2.repos)
- DDS implementation:
  - OpenSplice
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
C:\dev\ros2>set RMW_IMPLEMENTATION=rmw_fastrtps_cpp
C:\dev\ros2>ros2 run demo_nodes_cpp add_two_ints_server
C:\dev\ros2>ros2 daemon stop
The daemon has been stopped

C:\dev\ros2>set RMW_IMPLEMENTATION=rmw_opensplice_cpp
C:\dev\ros2>ros2 run demo_nodes_cpp add_two_ints_server
```  

```
C:\dev\ros2>set RMW_IMPLEMENTATION=rmw_fastrtps_cpp
C:\dev\ros2>ros2 node list
/add_two_ints_server
C:\dev\ros2>ros2 node info /add_two_ints_server
/add_two_ints_server
  Subscribers:
    /parameter_events: rcl_interfaces/ParameterEvent
 Publishers:
    /parameter_events: rcl_interfaces/ParameterEvent
  Services:
    /add_two_ints: example_interfaces/AddTwoInts
    /add_two_ints_server/describe_parameters: rcl_interfaces/DescribeParameters
    /add_two_ints_server/get_parameter_types: rcl_interfaces/GetParameterTypes
    /add_two_ints_server/get_parameters: rcl_interfaces/GetParameters
    /add_two_ints_server/list_parameters: rcl_interfaces/ListParameters
    /add_two_ints_server/set_parameters: rcl_interfaces/SetParameters 
    /add_two_ints_server/set_parameters_atomically: rcl_interfaces/SetParametersAtomically

C:\dev\ros2>set RMW_IMPLEMENTATION=rmw_opensplice_cpp
C:\dev\ros2>ros2 node list
/add_two_ints_server
C:\dev\ros2>ros2 node info /add_two_ints_server
/add_two_ints_server
  Subscribers:
  Publishers:
  Services: 
```

#### Expected behavior

I expect that the OpenSplice behavior matches the FastRTPS behavior (it displays the topics and services that are available.

#### Actual behavior

It doesn't.

#### Additional information

clalancette, 2018-12-14:
I noticed a similar thing with Connext.  In my case, the problem was that I had an "old" daemon still running (with the rmw_implementation set to fastrtps).  If I killed that daemon, then ran the talker with `RMW_IMPLEMENTATION=rmw_connext_cpp` and the `ros2 node info` also with the connext rmw, things were happier.  This still constitutes a bug in my opinion (or surprising user behavior, at least), but it would be interesting to know if that fixes it for you.

mjcarroll, 2018-12-14:
I explicitly killed the daemon, I included that in there as well.  I thought that may be the cause, but it doesn't seem to be.

clalancette, 2022-10-28:
We stopped supporting Opensplice a long time ago, so closing this out.

closed 2022-10-28 by clalancette