ros2/ros2cli #185: [ros2param] list misbehaves when hidden nodes are present

hidmic, 2018-12-14:
## Bug report

**Required Info:**
- Operating System:
  - Ubuntu Xenial 16.04 
- Installation type:
  - From sources
- Version or commit hash:
  - Crystal [ros2.repos](https://github.com/ros2/ros2/pull/624)
- DDS implementation:
  - ADLink OpenSplice 6.9 & eProsima Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

First, source your workspace and export an `$RMW_IMPLEMENTATION` env var for either vendor.

Run a hidden node. For instance;
```sh
ros2 run examples_rclcpp_minimal_action_server action_server_not_composable __node:=_action
```

Now, attempt to list parameters without specifying a node:
```sh
ros2 param list
```

#### Expected behavior
Nothing is shown (unless `--include-hidden-nodes` is given).

#### Actual behavior
In Fast-RTPS, CLI sometimes throws. In OpenSplice, CLI sometimes hangs.

clalancette, 2019-05-29:
This is still a problem on Opensplice with Dashing.

dirk-thomas, 2019-05-29:
This seems to be similar to ros2/ros2cli#246. The `action_server_not_composable` is not related to the problem. The hidden daemon doesn't have a parameter service and therefore the command hangs. See #265 for a fix.

clalancette, 2019-05-29:
> The hidden daemon doesn't have a parameter service and therefore the command hangs. See #265 for a fix.

I don't think that is true (or, at least, if it is, I don't understand why).

If I run `ros2 run examples_rclcpp_minimal_action_server action_server_not_composable` (without the node name remapping), and then run `ros2 node info /minimal_action_server`, I get a list of topics and services that includes `list_parameters`.  If then run `ros2 param list`, I get:

```
/minimal_action_server:
  use_sim_time
```

Which seems right to me.

If I add in the node name remapping with `ros2 run examples_rclcpp_minimal_action_server action_server_not_composable __node:=_action`, and run `ros2 node info /_action`, I also see `list_parameters` as one of the services.  However, if I run `ros2 param list` (even with the patch in #265), I get an empty response, which doesn't seem right.

dirk-thomas, 2019-05-29:
> The hidden daemon doesn't have a parameter service

The emphasis of my comment is on **the hidden daemon** - this has nothing to do with the action server example. Just try to call `ros2 param list --include-hidden-nodes` without running the example.

closed 2019-05-29 by dirk-thomas

jubeira, 2019-05-30:
This seems to be fixed in Dashing binaries for Ubuntu 18, both for Fast-RTPS and OpenSplice :+1:

PR #265 (2019-05-29): fix param list for nodes which don't have the service
Fixes #185.