ros2/ros2cli #191: ros2 topic pub does not seem to parse arrays of messages

mxgrey, 2019-01-25:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - 7a65e1ecf02c69bf3d9624d49b17cde57f7ed16a
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Run the following from the command line

```
$ ros2 topic pub some_pose_array geometry_msgs/PoseArray "{header: {stamp: {sec: 1, nanosec: 2}, frame_id: some_frame}, poses: [{position: {x: 0, y: 1, z: 2}, orientation: {x: 3, y: 4, z: 5, w: 6}}]}"
```

#### Expected behavior

The message gets published

#### Actual behavior

The console prints:

```
Failed to populate field 'poses': The 'poses' field must be a set or sequence and each value of type 'Pose'
```

#### Additional information

The specific example above isn't important, but it seems that `ros2 topic pub` cannot currently parse arrays of message types.

jubeira, 2019-02-14:
@cottsay would you please assign this one to me? Thanks in advance!

mikaelarguedas, 2019-02-17:
This looks like a duplicate of https://github.com/ros2/ros2cli/issues/59

I submitted a fix at https://github.com/ros2/ros2cli/pull/197 if you guys want to give it a try

mxgrey, 2019-02-17:
Sorry for duplicating the issue! You're right; it's the same problem. I'll close this to avoid duplicated tickets, and I'll test the fix at the first chance that I get.

closed 2019-02-17 by mxgrey

jubeira, 2019-02-21:
Just for the record, I tested the fix with #197 and https://github.com/ros2/rosidl_python/pull/30 and it worked fine :+1: