ros2/ros2cli #193: [ros2param] ros2 param list stuck if no node name provided

alsora, 2019-02-04:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  -binaried
- Version or commit hash:
  - crystal

#### Steps to reproduce issue
```
ros2 param list
```

#### Expected behavior

According to the help message of `ros2 param --help`, the previous command should display a list  of available parameters.
 
The fact that it requires a positional argument (the node name) is only specified if you run `ros2 param list --help`.

I would expect an error to be raised, saying that I need to specify one node name

#### Actual behavior

The process hangs up there, without saying anything.

dirk-thomas, 2019-02-04:
> The fact that it requires a positional argument (the node name) is only specified if you run `ros2 param list --help`.

That is incorrect the usage output mentions the node name in brackets:

> usage: ros2 param list ... [node_name]

This implies that passing a node name is optional. If you don't pass a node name the command will list the parameters of all nodes. If you do pass a parameter it only lists the parameters of that specific node.

If in you case the invocation is stuck that is likely due to some other problem. Please include more information how to reproduce the problem.

alsora, 2019-02-04:
I'm sorry, you are right.

I'm actually running Gazebo + Navigation stack and this is probably clogging my pc.
In this situation I also have problems setting/getting node parameters as I often get timeouts.

Tried again in a simpler scenario and everything works as expected.

closed 2019-02-04 by alsora