ros2/ros2cli #201: [ros2node] info fails to find nodes in namespaces

JacobJBeck, 2019-02-20:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - ros2cli 7a65e1ecf02c69bf3d9624d49b17cde57f7ed16a
  - rmw_fastrtps d8f1959a785ad770f96623d257b8a4ae1c74c938
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

1. Save script as `namespace.launch.py`
```
#!/usr/bin/env python3

from launch import LaunchDescription

from launch_ros.actions import Node

def generate_launch_description():
    talker1 = Node(package='demo_nodes_cpp', node_executable='talker')
    talker2 = Node(package='demo_nodes_cpp', node_executable='talker', node_namespace='ns')

    return LaunchDescription([talker1, talker2])
```

2. Run `ros2 launch namespace.launch.py`

3. Run `ros2 node info /talker`

4. Run `ros2 node info /ns/talker`

#### Expected behavior
Information is printed for both nodes.

#### Actual behavior
Node inside a namespace throws
```
[ERROR] [rmw_fastrtps_shared_cpp]: Unable to find GUID for node: talker
Failed to get_subscriber_names_and_types: Unable to find GUID for node , at /home/jacob/ros2_core_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_node_info_and_types.cpp:85
```

#### Additional information

cottsay, 2019-02-21:
I can reproduce this with the steps above. The given error is also different from when the node name is actually invalid, so something definitely seems off.

Thanks for the report.

ivanpauno, 2019-03-13:
@mjcarroll May if I take this one?

PR #206 (2019-03-15): Solved bug when trying to find node info in namespaces
Fixes #201 
Fixes #181 
Initial '/' was missing when the node was inside a namespace.
Now, non fully qualified names are also accepted.
Added unit test.
closed 2019-03-15 by ivanpauno