ros2/ros2cli #215: Add a CLI tool for Components.

hidmic, 2019-04-05:
## Feature request

Add a CLI tool to deal with the basic operations related to components and containers: listing, loading and unloading.

Connected to https://github.com/ros2/launch_ros/issues/6. 

#### Feature description

A rough sketch of the CLI is proposed below:

- list containers and their components:

   - show all containers with their loaded components:
     ```sh
     $ ros2 component list
     /some/component/container
         Components:
             4123753   /a_talker
             6232153   /a_listener
     /another/component/container
          Components:
                  2103   /my_action_server
     ```

   - show loaded components for a given container:

     ```sh
     $ ros2 component list /some/component/container
     4123753   /a_talker
     6232153   /a_listener
     ```

   - show component containers only
  
     ```sh
     $ ros2 component list --containers-only
     /some/component/container
     /another/component/container
     ```

- load a component into a container:
  - through command line arguments:

    ```sh
    $ ros2 component load /some/component/container rclcpp_components \ 
          test_rclcpp_components::Talker --node-name my_talker \
          --node-namespace /test --remap-rules chatter:=my_chatter \
          --parameters rate:=30 --parameters path/to/parameters/file.yaml \
          --extra-arguments intra_process_comm:=true
    ```

   - through files:

      ```sh
      $ ros2 component load /some/component/container -f path/to/components/listing.yaml
      ```   

      ```yaml
      # path/to/components/listing.yaml
      - package_name: rclcpp_components
        plugin_name: test_rclcpp_components::Talker
        node_name: my_talker
        node_namespace: /test
        remap_rules:
            /chatter: my_chatter
        parameters:
            - rate: 30
              ...
            - path/to/parameters/file.yaml
        extra_arguments:
           - intra_process_comm: true
      - ...
      ```

- unload a component from a container:

     ```sh
    $ ros2 component unload /some/component/container 4123753
    ```

- list all available components:

    ```sh
    $ ros2 component types
    rclcpp_components   test_rclcpp_components::Talker
    rclcpp_components   test_rclcpp_components::Listener
    ```

dirk-thomas, 2019-04-05:
LGTM.

I would suggest to leave the `-f <yaml-file>` option out for now. It is yet another thing to document and not really necessary.

mjcarroll, 2019-04-05:
LGTM!

PR #217 (2019-04-15): Add Component CLI
Closes #215. ~Still under test.~ Tested locally.
closed 2019-04-15 by hidmic