ros2/ros2cli #219: [ros2action] Calling cancel even after successful execution

Karsten1987, 2019-04-15:
## Bug report

**Required Info:**

- Operating System:
  - OSX
- Installation type:
  - source
- Version or commit hash:
  - latest master

#### Steps to reproduce issue
```
ros2 action send_goal --feedback <action name> <action type> "<data>"
```

#### Expected behavior
```
Goal finished with status: SUCCEEDED
```

#### Actual behavior
```
Goal finished with status: SUCCEEDED
Canceling goal...
Failed to cancel goal
```

#### Additional information
My believe is I see `Failed to cancel goal` because my implementation of the action server does not handle a cancel request atm.

## Feature request
When goal is successfully reached, don't call cancel.

jacobperron, 2019-04-16:
I'm unable to reproduce this. If you could post a SSCCE for the action server that would be helpful.

Karsten1987, 2019-04-16:
hmm, maybe an OSX thing?

```
 ➭ ros2 run examples_rclcpp_minimal_action_server action_server_not_composable
[INFO] [server]: Got goal request with order 10
[INFO] [server]: Executing goal
[INFO] [server]: Publish Feedback
[INFO] [server]: Publish Feedback
[INFO] [server]: Publish Feedback
[INFO] [server]: Publish Feedback
[INFO] [server]: Publish Feedback
[INFO] [server]: Publish Feedback
[INFO] [server]: Publish Feedback
[INFO] [server]: Publish Feedback
[INFO] [server]: Publish Feedback
[INFO] [server]: Goal Suceeded
```

```
 ➭ ros2 action send_goal --feedback /fibonacci example_interfaces/Fibonacci "{order: 10}"
Waiting for an action server to become available...
Sending goal:
     order: 10

Goal accepted with ID: 06c856ccc5ba4c16b18f8bf432d68f23

Feedback:
    sequence: [0, 1, 1]

Feedback:
    sequence: [0, 1, 1, 2]

Feedback:
    sequence: [0, 1, 1, 2, 3]

Feedback:
    sequence: [0, 1, 1, 2, 3, 5]

Feedback:
    sequence: [0, 1, 1, 2, 3, 5, 8]

Feedback:
    sequence: [0, 1, 1, 2, 3, 5, 8, 13]

Feedback:
    sequence: [0, 1, 1, 2, 3, 5, 8, 13, 21]

Feedback:
    sequence: [0, 1, 1, 2, 3, 5, 8, 13, 21, 34]

Feedback:
    sequence: [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55]

Result:
    sequence: [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55]

Goal finished with status: SUCCEEDED
Canceling goal...
Failed to cancel goal
```

jubeira, 2019-04-16:
I could reproduce this with:
- Ubuntu 18.04
- Version 0.7.0 from debians (Dashing pre-release)
- `fastrtps` and `connext`; **not** with `opensplice`.

It doesn't happen always (I'd say ~20% of the times). Same output as @Karsten1987 described.

Karsten1987, 2019-04-25:
@jacobperron: This is still happening for me at every command I issue.

PR #254 (2019-05-23): reset goal_handle to avoid attempt to cancel
Fixes #219.
closed 2019-05-23 by dirk-thomas