ros2/ros2cli #221: [ros2component] List occasionally fails to retrieve components

jubeira, 2019-04-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - Hash `930dea5c8de47d47ac0b3814ee1c08bac8a70c22` (`Fix component arguments to be inline with service. (#220)`) + `rclcpp_components` from binaries version `0.7.0`.
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
1. From a terminal, run `rclcpp_components`:
```
ros2 run rclcpp_components component_container
```
1. From a second terminal, add some components (tested with 2 talkers and 2 listeners):
```
ros2 component load /ComponentManager composition composition::Talker
ros2 component load /ComponentManager composition composition::Listener
ros2 component load /ComponentManager composition composition::Talker
ros2 component load /ComponentManager composition composition::Listener
```
The interaction between talkers and listeners should be displayed in the first terminal, e.g:
```
[INFO] [talker]: Publishing: 'Hello World: 114'
[INFO] [listener]: I heard: [Hello World: 114]
[INFO] [listener]: I heard: [Hello World: 114]
[INFO] [talker]: Publishing: 'Hello World: 29'
[INFO] [listener]: I heard: [Hello World: 29]
[INFO] [listener]: I heard: [Hello World: 29]
```
1. List the components:
```
ros2 component list
```
#### Expected behavior
Components should be listed like this:
```
/ComponentManager
    1  /talker
    2  /listener
    3  /talker
    4  /listener
```
#### Actual behavior
~20% of the times, no component is listed.

#### Additional information
This doesn't seem to happen in `fastrtps` nor `opensplice`.

dirk-thomas, 2019-05-22:
Does the problem disappear when passing `--spin-time N` with a long enough period to ensure the node started by `ros2 component list` has enough time to complete discovery?

jubeira, 2019-05-22:
I need to rebuild by workspace with the current packages to retest properly; right now it is heppening to me even with `--spin-time 10`, but I want to make sure it's not a local issue. I'll try again and post the results here.

jubeira, 2019-05-23:
@dirk-thomas I can't reproduce this issue anymore using an up-to-date workspace from source.
For completion I tried listing the components 20 times in a row with `FastRTPS`, `OpenSplice` and `Connext` and it didn't fail even a single time. `--spin-time N` option wasn't required.

I think this issue is safe to close.

closed 2019-05-23 by jubeira