ros2/ros2cli #222: [ros2action] Display info about active action goals on an action server

jacobperron, 2019-04-17:
## Feature request

#### Feature description

Proposed originally in https://github.com/ros2/ros2cli/issues/202.

Add a `goal` verb for getting info about active or terminated goals for a given action server.

Proposed usage:

- List goals for a given action name. Output node names with action servers and goal IDs.

```sh
ros2 action goal list [--count] ACTION_NAME
```

E.g.

```sh
$ ros2 action goal list /fibonacci
/node/foo: 2
    152790028afb49a3b3c61383e6b97320
    8c3449e71c9941ed8ad3d68fdd7f553c
/another/node/bar: 1
    b07cb18ca2dc4cca95acfc131fc49223
```

```sh
$ ros2 action goal list --count /fibonacci
/node/foo: 2
/another/node/bar: 1
```

- Get details for a specific goal (request, result, status).

```sh
ros2 action goal info [--goal-only] [--status-only] [--result-only] ACTION_NAME GOAL_UUID
```

```sh
$ ros2 action goal info /fibonacci 152790028afb49a3b3c61383e6b97320
Goal:
  order: 5

Status: EXECUTING
```

```sh
$ ros2 action goal info /fibonacci 8c3449e71c9941ed8ad3d68fdd7f553c
Goal:
  order: 5

Status: SUCCEEDED

Result:
  sequence: [0, 1, 1, 2, 3]
```

#### Implementation considerations

I believe most of the proposed API should be feasible, except for accessing the goal request arguments, which requires adding some new API for getting this information to the `rcl` level.

jacobperron, 2019-04-17:
Alternatively, [it has been proposed](https://github.com/ros2/ros2cli/issues/202#issuecomment-472696112) to instead merge this feature with the `info` verb.

E.g

```sh
$ ros2 action info --goal-request-only /fibonacci
/node/foo: 2
  152790028afb49a3b3c61383e6b97320
    order: 10
  8c3449e71c9941ed8ad3d68fdd7f553c
    order: 1
/another/node/bar: 1
  b07cb18ca2dc4cca95acfc131fc49223
    order: 5
```

ste93, 2022-02-09:
Has anyone got any news from this?

clalancette, 2022-02-09:
We're not currently working on this feature, but we are happy to review pull requests that implement it.