ros2/ros2cli #223: Add ros2interface

jacobperron, 2019-04-17:
## Feature request

#### Feature description

Originally proposed in https://github.com/ros2/ros2cli/issues/202#issuecomment-472540339.

Replace `ros2 msg` and `ros2 srv` commands with `ros2 interface`, which also can handle action definitions (and possible more general IDL definitions).

Example usage:

```sh
# List all .msg, .srv, and .action types available
 $ ros2 interface list
 Messages:
     # Message types listed here
 Services:
     # Service types listed here
 Actions:
     # Action types listed here

 # Only list .msg types
 $ ros2 interface list --only-msg
 Messages:
     # Message types listed here

 # List all packages that generate messages, services, or actions
 $ ros2 interface packages
 action_msgs
 builtin_interfaces
 diagnostic_msgs
 # etc ...

 # Only list packages that generate actions
 $ ros2 interface packages --only-actions
 action_msgs
 example_interfaces
 test_msgs
 # etc ...
 
 # Show an interface definition
 $ ros2 interface show example_interfaces/Fibonacci
 Action:
     # Goal
     int32 order
     ---
     # Result
     int32[] sequence
     ---
     # Feedback
     int32[] sequence
 
 # In case of ambiguity (e.g. message and service with same name), show both
 $ ros2 interface show test_msgs/Primitives
 Message:
     bool bool_value
     byte byte_value
     char char_value
     # etc ...
 Service:
     bool bool_value
     byte byte_value
     char char_value
     # etc ...
     ---
     bool bool_value
     byte byte_value
     char char_value
     # etc ...
```

artivis, 2019-06-28:
After the merge of #288, will `ros2 msg` and `ros2 srv` be deprecated or are they all going to live side by side ?

jacobperron, 2019-06-28:
> After the merge of #288, will ros2 msg and ros2 srv be deprecated or are they all going to live side by side ?

IMO they could be deprecated for Eloquent, but open for discussion.

jacobperron, 2019-06-28:
Closed by #288

closed 2019-06-28 by jacobperron